cp2004/OctoPrint-WS281x_LED_Status #2: [Feature Request] Allow setting to solid white printing to light up the print area

tonesto7, 2020-07-28:
###

cp2004, 2020-07-29:
@tonesto7 Thanks, this is a great idea. It's on my list, would a good implementation be a navbar button that lights up for ~20secs maybe?

Hypertoken, 2020-07-30:
I think he just means during the entire print. Not sure, that was what I was looking for. I just set the progress to indicate solid white, and background to solid white. However, I cannot seem to get it to display while printing anything. Heater progress worked, but not printing progress. I think there might be another plugin causing the issue. I'm going to check that.

cp2004, 2020-07-30:
Ah Ok I see now. I had another request from somewhere that was similar. Will see what I can think of. Maybe a toggle between having progress effect on or one of the standard effects, then it can be what you like

cp2004, 2020-08-02:
Feature ready for 0.4.0. For anyone following this, (once released) enable the 'printing' effect and set to solid white (Should be default) for this. Will override Print Progress effect.

daleward, 2020-08-02:
Would love  to have it go solid bright light for as a flash for time lapse photo. Would also if you could make a gpio input toggle light brite white for working on it or back to status.

cp2004, 2020-08-03:
@daleward Would you mind opening a new request please, it's related to this one but I have already implemented a setting for solid white, and would like to release & close this one. Thanks.

cp2004, 2020-08-03:
0.4.0 has been released.

closed 2020-08-03 by cp2004