cp2004/OctoPrint-WS281x_LED_Status #3: I can't connect a WS2812B led strip

AMJordan886, 2020-07-29:
Hello, as the title says, I can not turn on the WS2812B led strip that I have. As adafruit says I have connected the strip with external 5V power and in turn the strip to GND and Pin10 (19) for signal and does not light. Is the connection scheme wrong? Should I put a resistance as other forms of connection show? thank you very much for the great work

cp2004, 2020-07-29:
Couple of things to check for you:
1. Did you complete the setup wizard fully (It shouldn't allow you to miss it!), so SPI is enabled.
2. Have you made sure the settings are correct on the edit strip settings pop-up? There is an option for WS281B, It should work as someone volunteered to test with WS2812. 
3. If you still get nothing, can you enable custom debug logging (On the plugin's about page) and send the plugin_ws281x_led_status_debug.log my way. (Give it some time to fill with some info just after enabling)

Thanks

michaelwilkinson1992, 2020-08-03:
I have a WS2812 8 LED strip. I have 5v, GND and GPIO attached directly to the Pi and can confirm it works.

cp2004, 2020-08-05:
@AMJordan886 Any update on this? If I don't hear anything in another 7 days this ticket will be closed.

cp2004, 2020-08-09:
@AMJordan886 You've got 3 days.... I'm happy to help, but can't if I don't hear anything!

cp2004, 2020-08-12:
8 days later, closing.

closed 2020-08-12 by cp2004