cp2004/OctoPrint-WS281x_LED_Status #5: [Feature Request] Intercept M150 command

Hypertoken, 2020-07-30:
Intercept M150 command; Set RGB(W) Color. Similar to the LEDStripControl plugin, but for Neopixels.

cp2004, 2020-08-01:
@Hypertoken Just looking into this now: May not make it in for a bit as I have a few things on the list for next version already. Shouldn't be too hard, can fire it at the effect handler easy enough. Would mean on the next OctoPrint event, it would go to whatever effect but they could be easily disabled. And as it stands it would only be RGB color, W would be ignored.

Hypertoken, 2020-08-01:
Ok, can't wait! Just a thought on handling "W" . If "W" represents white, and equal RGB values also represent intensity of white. Something like this might be a work-around; 

if (W>0) Then { 
     R += W;
     G += W; 
     B += W;
} 

OR for even more fine tune of color and intensity of white; 

if (W>0) Then { 
     R += W;
     if (R > 255) Then R = 255;
     G += W;
     if (G > 255) Then G = 255;
     B += W;
     if (B > 255) Then B = 255;
} 

But this probably won't work, because I suck a programming!

cp2004, 2020-08-02:
Had another idea for this... The W could become brightness setting, if enabled. Adding to the 0.5.0 milestone as there's a few things ahead of this.

cp2004, 2020-08-03:
Implemented ready for 0.4.0. Needs a setttings page, but functionally there. Will write a Wiki entry with details before release.

cp2004, 2020-08-03:
0.4.0 has been released

closed 2020-08-03 by cp2004