cp2004/OctoPrint-WS281x_LED_Status #6: [Feature Request] A way to add ON/OFF to the octoremote Android

salilgupta4, 2020-08-02:
Is there a way to add the button to the Octoremote so that the Lights can be switched ON or OFF from the app itself. Even though the timed option is there. But sometimes just doesnt help. 

Thanks,

cp2004, 2020-08-02:
@salilgupta4 Can you tag the OctoRemote developer here? The lights on/off can be accessed from the API but as far as I know there isn't a way to add custom API commands there.

cp2004, 2020-08-02:
Not sure if the app is open source, but you may be able to add a custom control. I can take a look at some point, but would need to work out how they work first.

salilgupta4, 2020-08-02:
The developer is Octoremote itself. 
[image]
So here in the description they have mentioned that we can add some controls. Just like you said, the function has to be called, but how?
[image]

This is no major issue. But can spread its wings to android too. And actually the Octoremote app is pretty handy.

Thanks for looking into it.

cp2004, 2020-08-03:
@salilgupta4 Solution for you! 
In 0.4.0 I have added 2 @ commands, `@WS_LIGHTSON` and `@WS_LIGHTSOFF`

These can be sent in the gcode stream to OctoPrint, picked up and interpreted.

To get them into OctoRemote, head to Settings > Controls >Edit custom controls. Then add command, in the `commands` box you can put either of the above. Once saved, they are on the control tab in OctoRemote

salilgupta4, 2020-08-04:
Thats a big help. Thanks so much. 

No many people provide solutions that quickly.

cp2004, 2020-08-04:
The @ commands were on my list already, but got bumped forwards when I noticed it would solve this... Glad to know it helped!

closed 2020-08-04 by cp2004

gortador, 2021-09-17:
Is there a possibility of being able to add the effects via commands? Gcode ? or @WS_RAINBOW_CYCLE ? or other ?

7R0J4Ner, 2021-09-20:
@cp2004 would it be Possible to add something like `@WS_LIGHTSTOGGLE` ?
I dont like to have 2 Buttons to Controll my Lights.