cp2004/OctoPrint-WS281x_LED_Status #8: [Feature Request]

cp2004, 2020-08-03:
From comment in #2 @daleward
> Would love to have it go solid bright light for as a flash for time lapse photo. Would also if you could make a gpio input toggle light brite white for working on it or back to status.

* Solid bright light for flash for time lapse

* GPIO toggle for lighting white while working.

This is more challenging. The GPIO toggle should be doable, the timelapse one may be harder as it is dependent on more precise timing, and there is not a way to know in advance a timelapse snapshot will be taken, only at start/end

daleward, 2020-08-04:
The most common time lapse in Octoprint,  I think would be take the photo on  layer change

ssl-3, 2020-08-06:
> The most common time lapse in Octoprint, I think would be take the photo on layer change

You're right, I think.  And the layer change is, I think, predictable from within Octoprint.

And I'm just spitballing here, but:  Unless the camera is manually-adjustable (it doesn't seem to be by default), then...  it'll take a bit (several seconds, perhaps) for the camera to adjust to the new influx of light.

So while just going _flash_ at the right instant will certainly produce a bright image, it will also certainly result in a very overexposed image in any (dark-ish) environment where the flash effect would be useful to begin with.

I'd like to see this feature implemented, too, but I think that it is a much deeper problem than one might initially assume.

cp2004, 2020-08-11:
Going to be implementing a 'torch' kind of mode, with either a button from the navbar or GPIO, then stays white for x seconds

cp2004, 2020-08-19:
0.5.0 Has been released with the torch mode. GPIO coming soon.

cp2004, 2020-10-18:
I'm going to put this issue up for adoption, since I don't have time to implement it myself. If anyone from the community would like to take this on, let me know!

cp2004, 2020-12-06:
Note for this one: It should be done as a separate script/plugin making an API call to this one. I don't want to be dealing with more GPIO here. If anyone wants to, feel free.

roboraptor, 2021-01-06:
Having a pin would be awesome! I can imagine coming to printer at night and light it up on press of a button.

> 
> 
> The most common time lapse in Octoprint, I think would be take the photo on layer change

If you are using basic timelapse which is taking picture on every layer, 
You can add @WS_TORCH to your slicer to add it on every layer change.
It will light up for a time which you can set in plugin settings and then you can take nice picture

cp2004, 2021-01-06:
That's exactly it, this was kind of added to the wiki here: https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Tips-and-Tricks#timelapse-flash, though it's probably not easy to find. If I add the GPIO script, it will become more of a central place for things like this.

I haven't got round to writing a script that would do this, but it would be very easy to do with a simple API call to the plugin. Which also means the API should be documented somewhere 😉 

I'm not going to put any more GPIO based logic in this plugin, it causes way too many headaches. For example, if people use filament sensors, or PSU control, it often causes conflicts. By creating a quick (probably Python) script that you can download and run, it means I don't have to worry about any problems with this.

cp2004, 2021-02-08:
Closing this issue, in 0.7.0 there will be a stable API that will be [documented](https://cp2004.gitbook.io/ws281x-led-status/documentation/rest-api) that users can use to create GPIO scripts etc.

No one stepped up to provide a script, and it won't be implemented in the core plugin because I don't want to mess with more GPIO stuff at the same time as LEDs. If someone still wants to write it, I would be happy to add it to the [guides section of the docs](https://cp2004.gitbook.io/ws281x-led-status/) - or it could be done with a PR.

closed 2021-02-08 by cp2004