cp2004/OctoPrint-WS281x_LED_Status #9: WS8212B - leds keep blinking

albertgn, 2020-08-03:
**What's your question?**

I run Octoprint on a Raspberry Pi 3b

I installed the plugin in Octoprint v1.4.0.  I Followed the setup wizard and all worked fine. 

The WS8212b strip has a separate power supply. The Raspberry PI has also a separate power supply. They share the common.

Data line is on GPIO 10 . 

I am using a 3.3 - 5V logic level shifter because without it the leds would not light at all.

No matter how many leds i choose from the plugin setup (I reduced the count to 10) I have the same situation. The leds on the strip will keep blinking (on and off all the time). 

During the printing process, the effects seem to be working but with the blinking at the same time. For example the bounce will work but the three leds will be blinking too. The solid color effect will be like 10 leds blinking in the chosen color.

I am on the latest version (0.4.0) of the plugin. The previous version had the same behavior.

Here is what I see in the log :
----
[2020-08-03 22:42:57,875] INFO: Initialising LED strip
[2020-08-03 22:42:57,878] INFO: Strip object successfully initialised
[2020-08-03 22:42:57,879] INFO: On message recieved, turning on LEDs
-----

So before I log an issue, I wanted to ask if there is something else in the configuration of the plugin that I should do ?

cp2004, 2020-08-04:
This.... is an interesting one. Not something I've come across at all, and haven't got much of an idea how to solve it. But we'll give it a go. Here's what I'm thinking

1. I don't think this is an issue with the plugin turning the LEDs on or off, as the logs indicated they are turned on at startup, as you would expect. 

2. When you say:
    >They share the common.

    I'm going to assume you mean the ground connection. Not sharing ground pins has caused quite a lot of flickering for me previously

3. You mention you have a separate power supply, which is what I would say if you had them connected direct to the Pi.... Do you have a way to confirm the power supply is working well?

4. Do you have a way of testing the strip outside of the plugin, to narrow down if it is hardware or software? You could try this maybe? https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/

I'm inclined to think it may be a hardware issue, my gut feeling is power. Let me know what you think of the above

albertgn, 2020-08-04:
Thanks for the reply. Yes by common I meant the ground.

The strip is working well. I tried it with Arduino code and run the examples that come with the FastLED library. All good.

Now the power supplies are in good shape in my opinion, but this is something I will double check. I have a separate power supply for the strip to make sure they get the needed current. In Theory with 10 LEDs they should draw around 600mA. I put a power supply that can give 2.5 A and the intention is to get 25 LEDs installed on the printer, so I think that should be fine.

I too think it is hardware related, but wanted to see if I was missing some configs.

The solid color effect is supposed to keep the LEDs on permanently , right ?

cp2004, 2020-08-04:
Yes, you are correct, solid colour is always-on. So it's not an issue with the LED strip or power supply, 2.5A should be easily up to it (I have measured it in the past, and you rarely hit the max calculated power)

Let me have a look at the FastLED library, and see if there is some settings in there that might help for you. Other than that, I think the only thing that it could be is using the Raspberry Pi SPI drive, rather than Arduino using PWM.

albertgn, 2020-08-04:
sorry another question : did you use any logic converter or you directly connected the GPIO to the DIN of the strip ?
In my case I couldn't have any action on the strip when directly connected to the Raspberry Pi. For this reason I added in between a 3.3 to 5V logic level shifter. Tested it on other circuits and works fine.

cp2004, 2020-08-04:
I do have a level shifter in my circuit, I soldered together a little add-on for the Pi. 

Couldn't make my way through FastLED library, and I couldn't find anything about this issue anywhere else. Would you be able to SSH into the Pi with OctoPrint on and test it out - 

If you are on OctoPi run:
1. `source /home/pi/oprint/bin/activate` (Activates OctoPrint's virtual environment)
2. `curl https://github.com/rpi-ws281x/rpi-ws281x-python/raw/master/examples/strandtest.py --output strandtest.py` (Downloads the example script from library)
3. `nano strandtest.py` and change `LED_PIN` to be pin 10 (`LED_PIN = 10`)
4. Run the script with `python strandtest.py`

That runs some effects, if it doesn't work that at least means it's not my fault :) Narrows it down to (probably) the library or the strip not liking the SPI.

To test if it is SPI you can try:
1. Plug LED data wire to Pin 18 (PWM)
2. Change the script to use `LED_PIN = 18`
3. Run using `sudo python strandtest.py`

Lets hope we can get this at some point

albertgn, 2020-08-04:
I got it !...it is as you mentioned earlier : the power supply.
 I changed it with another one and everything worked. 

I need to check it better to understand the reason, because at this moment the only difference I see between the two power supplies is that the new one provide 4A of current. I prefer to think the old one is just faulty than think the 10 LEDs need 4 amps to run :-)

btw I am connected on GPIO 10....GPIO 18 can't have access without root (both plugin and the script you provided)

Thank you for taking the time answering and helping me

cp2004, 2020-08-04:
Great that you got it! Power supplies can just... Die sometimes. Enjoy the plugin!

closed 2020-08-05 by cp2004