cp2004/OctoPrint-WS281x_LED_Status #10: [Feature Request]

m00x-pixel, 2020-08-04:
Maybe it can be possible to work with the WLED API from Aircookie via UDP wireless connection.
(ESP8266/ESP32 with connected LEDs)

So you can can have an small printer status light within the living room and see the state without going to the printer in the other room.

cp2004, 2020-08-04:
Interesting idea... I kind of feel like it would get too complicated to integrate this with this plugin, and would be more suitable for it's own plugin. Shouldn't be too hard for me to replace the effect runner class in this plugin to a different one that sends HTTP requests.

I had never heard of Aircookie's WLED before, looks nice. I haven't got time now, but will investigate.

cp2004, 2020-08-04:
Just seen I can pick up an ESP8266 board for £4.99 👀

m00x-pixel, 2020-08-04:
It would be nice if it will be possible. Thanks to look into.

Yes, the ESP8266 are really cheap. I using already many of them with lights and backlight for TV using UDP and hyperion server

saikek, 2020-10-19:
It also would be possible to send data via MQTT. Then Tasmota devieces can also work.

Prutsium, 2020-10-24:
Just wanted to create the same request.
The advantage to be able to use WLED is that the LED and printer could be seperate from each other as there would be a wireless connection.

In my case i would use Nanoleaf (clone running WLED) thats in my living room.

cp2004, 2020-10-26:
You'll all be pleased to know I've started research into developing this plugin. 🙂 
It'll take a while, but I have some ideas of how to go about it. Will ping you here when I have a beta kind of version to get some feedback, it'll be a new plugin.

saikek, 2020-10-27:
Check this guys:
https://github.com/Aircoookie/WLED

Probably something can be used as an example.
https://github.com/Aircoookie/WLED/blob/f66c28a1bca36ced44666621e0b4f2b3fcc7a8fa/wled00/mqtt.cpp