cp2004/OctoPrint-WS281x_LED_Status #12: Can't find plugin after installed and enabled.

ssl-3, 2020-08-06:
I installed and enabled the plugin from Octoprint's Plugin Manager, but can't find it anywhere.  I've restarted Octoprint.

Perhaps I'm just looking in the wrong places?  The documentation seems to assume that I know how to start it, and I thought that I would (I certainly have other plugins doing their plugin-stuff).

I expected it to show up under the Wrench menu under Plugins, but there's nothing there related to WS281x.

What am I doing wrong here?

cp2004, 2020-08-06:
Strange. Once installed OctoPrint should start it automatically. Couple of things to ask:
1. Any errors during install?
2. Can you upload an `octoprint.log` of when you installed the plugin & restarted OctoPrint, provides some useful things about the environment.

ssl-3, 2020-08-06:
No reported errors during install.  The only prompts were to reboot/reset after installing, and to reboot/reset after enabling.

I can get the log file for you later if you'd like.  I'm tired right now, and scp is failing me.

cp2004, 2020-08-06:
I can wait :) You can also just download them from OctoPrint's logging tab (Settings > Server > Logging) if scp is not going well for you.

ssl-3, 2020-08-06:
A log for your perusal.  :)

[octoprint (1).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5038490/octoprint.1.log)

ssl-3, 2020-08-07:
Well, that's sure interesting:  Progress.

I connected Octoprint with Cura, and when Cura's Octoprint plugin did an API key request to Octoprint, the configuration wizard for the WS281x plugin came up as if by magic.

Just like -- I presume -- it should have done so to begin with.

It presumably even works (I haven't wired anything up yet).  I've got a new button with a lightbulb on my top bar.  Software-wise, things appear to be A-OK so far.

I'm attaching another octoprint.log that contains the success story and leaving the issue open for now, because what I was experiencing before was certainly very strange and I have no idea how to escape from that.
[octoprint (2).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5040211/octoprint.2.log)

Let me know what additional troubleshooting you'd like me to do.  I'm even willing to try to replicate the issue with a fresh copy of Octopi, if that is useful.

cp2004, 2020-08-07:
All through the log, it seems to think the plugin was enabled & working. I can see the point it was installed, successfully and then restarted and the plugin is there, logging away when it starts and stops. I also 

My guess is that it could be a browser/caching issue, where it would have needed to fully reload the web interface to get the plugin pages to come up - but I can't tell if that happened from the logging, only if it's on debug level, then you've got 1000s of things to fish through. Can you remember if you were prompted for a reload by OctoPrint after you installed & restarted?

Maybe I'll give that a go, see if I can get it to install but not refresh page. Seems the most likely thing, given it was there in the backend

ssl-3, 2020-08-07:
I was prompted to restart Octoprint immediately after installing the
plugin, but I do not recall being asked to do any special browser-reloads.

It may be a browser caching issue -- I think you're on the right track with
that.  The Cura Octoprint plugin spawned a new browser instance.

And I definitely did not try any browser resets while [attempting to be]
configuring your plugin -- I did it in the same Octoprint tab (in Chrome,
on Windows) that I had open for days and days.

If that's all it is, then a small documentation change ("close and re-open
your browser to prevent weirdness") would make it fixed enough that I'll
not personally run into the problem again ever.  That'd be good enough for
me.  :)

cp2004, 2020-08-07:
That does clear up the problems then. Any plugin that registers a wizard, or a template to inject (as this does) should trigger the 'reload' prompt, through `octoprint.plugin.ReloadNeedingPlugin`. Can add the bit to the readme, and maybe give it a bit of a test to make sure it's doing what I think it should be.

cp2004, 2020-08-08:
Have just updated the readme & wiki with notes about this. (See above commit) It seems that in rare cases, OctoPrint doesn't trigger a reload on server restart, so it *could* be a bug in OctoPrint. Unsure still, but might be worth me looking into. The plugin explicitly subclasses `octoprint.plugin.RestartNeedingPlugin` as well as `octoprint.plugin.ReloadNeedingPlugin` through `AssetPlugin` and `TemplatePlugin`, however my suspicion is that the restart takes precedence and the reload is forgotten on restart.

ssl-3, 2020-08-11:
I agree that it may be an Octoprint bug.  I certainly followed whatever
prompts it gave me before winding up with the inconsistent behavior
described above.

And I haven't looked at your changes to the documentation, but:  Thanks!
I'm sure that is is cromulent.

Feel free to forward to me your your Paypal address.  I'd like to toss you
a few bones.

cp2004, 2020-08-12:
In the process of setting up Github sponsors and Paypal etc, will let you know once I have something ready

cp2004, 2020-08-18:
GitHub sponsors is now live, at https://github.com/sponsors/cp2004

Going to close this as it has been solved, and it's on my list for my install tests for 0.5 to check the OctoPrint bug out.

closed 2020-08-18 by cp2004