cp2004/OctoPrint-WS281x_LED_Status #14: [Feature Request] Reverse Order on Progress

11chrisadams11, 2020-08-11:
It would be nice to be able to reverse the light up order on the progress effect. I have my strip running horizontally, and the way I have it wired, it lights up from right to left. It would be cool to have an option to reverse it in the settings.

Is this something that is in the works? If not, I could take a look at implementing it.

Also, is there a command that I can run to test out the progress without actually running a print?

Thanks

cp2004, 2020-08-11:
This one's on my list but wasn't close to the top, as no one had requested. So you're welcome to go for it!

Couple of things:
* Make sure you update your fork from the dev branch, as there are some more effects & changes that have gone in recently
* You might be able to make use of `for i in reversed(range(strip.numPixels()))` so it moves in reverse. I already do this in some of the effects using the two 'directions', like here, so the effect goes 'out and back'. Might be able to do something similar, adding an optional arg to the effects, which can then be passed through from the runner.
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/blob/30103453b26801802b68155d3c234f375dacb9e0/octoprint_ws281x_led_status/effects/basic.py#L37-L47
* WRT testing, I do all of my testing against the virtual printer, so get yourself at least OctoPrint 1.4.1, and you can enable it's plugin from the settings page - should then be a new serial port 'VIRTUAL'. I have a couple of short (between 1-5mins, VP can go quite fast) files that I run.

cp2004, 2020-08-11:
@11chrisadams11 Just thought about this a little bit more, it would be easiest to add an arg to the effect method, so you have `strip, queue, value, progress_color, base_color, max_brightness=255, reversed=False`
then you can query which way to do the effect. Then in `runner.py` & `__init__.py`, it can be added as an option - if you can't make sense of how the settings go across, I can help out.

Might be something to consider doing to all effects, but start with progress.

11chrisadams11, 2020-08-11:
Great. Thanks for the pointers. I will look into it. I was thinking about making it an argument to make it universal in the future.
I will let you know if I run into any problems.

Thanks

cp2004, 2020-08-17:
@11chrisadams11 Hey! I've added a 'contributors' section to the about page now. Would you like to be know as 11chrisadams11 or an actual name (Let me guess... Chris Adams?). Don't mind if you don't want either, but thought it would be nice if to credit.

11chrisadams11, 2020-08-18:
Just the username is good. Thanks for including me in that.

cp2004, 2020-09-09:
Hi @11chrisadams11 

Just thought I'd check in to see if this was in-the-works, or if you've got a particular timeframe of when you might look at it 👀 ?

I'm working on more ways progress functions are used now (cooling tracking), so would be interesting if we could get a 'reverse' option tied in with a release for more progress functions.

No worries if you're no longer looking at this, priorities change, I just need to know to plan when/what to release. I'm still happy others have contributed 🙂

11chrisadams11, 2020-09-10:
Yes, I did start work on it. I have the reverse light up for the progress effect working. I am trying to figure out how to tie it to the front end and settings. 

I have been crazy busy lately, so I haven't had much time to work on it much.

cp2004, 2020-09-10:
@11chrisadams11 Cool, thanks for letting me know. I've been busy myself, know what you mean with that! Its hard to find time for these nice things :)

cp2004, 2020-10-25:
UI setting added, ready for 0.6.0

cp2004, 2020-10-29:
@11chrisadams11 could you test 0.6.0-rc.1 with this feature in it?

Can be installed by using the following URL in plugin manager > Get more > ...from URL:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

Please provide general feedback in the [feedback ticket](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/49) - an 'all is working fine' is valuable feedback since it lets me know people are testing this release. Thanks!

11chrisadams11, 2020-10-29:
I have tested it, and it works great. I tried normal and reversed and both worked. It even works when switching in the middle of a print.

Thanks for the update.

commit 2bde1e34 (2020-10-25): :sparkles: Add reverse progress setting to UI, & assorted UI updates

* New setting reverse, under 'Progress' on the UI.
* <hr> between settings of the same type
* Added sponsors modal

Closes #14, linking #37 as well.
closed 2020-11-01 by cp2004