cp2004/OctoPrint-WS281x_LED_Status #15: [Feature Request] Adjusting brightness of LEDs

fnsign, 2020-08-12:
I am about to build an enclosure for my Ender 3 and found this plugin which would absolutely meet my needs and make me get away from the firmware option in Marlin for Neopixels.

I haven't installed the plugin yet, but I read through the doc. I couldn't find an option to adjust the brightness of the LEDs. That would be a very useful feature, as WS281x tend to draw quite some power and get rather hot, esp. when running them in full white. By adjusting the brightness that could be overcome at some extent. The illumination in my case will be a strip with 60LEDs (1m).

Is that feature planned for a future release?

Cheers!

cp2004, 2020-08-12:
AHH! There's a brightness setting :) it is in the strip settings, but just realised it's not in the docs!

Will sort this out ASAP!

cp2004, 2020-08-12:
Record for fastest request to be solved...

fnsign, 2020-08-13:
Absolutely fantastic! Thank you!

closed 2020-08-13 by fnsign