cp2004/OctoPrint-WS281x_LED_Status #16: [Feature Request] Temperature monitoring after printing / while hot

ssl-3, 2020-08-17:
**Is your feature request related to a problem? Please describe.**

Problem:  People sometimes assume the printer is cooled down just because it has finished a print.  This results in "ouch!" more than it could.

**Describe the solution you'd like**

It'd be more-funnerer if the plugin monitored temps at idle and would illuminate a user-chosen pattern or color if temp of "item X" > "temperature Y".

**Describe alternatives you've considered**

Yelling louder to pay attention after I hear "ouch!".

cp2004, 2020-08-17:
Something I could definitely look at, putting it on the roadmap for 0.6.0, as I want to get 0.5 out this week ideally and still have a few bits left to do on that one.

ssl-3, 2020-08-17:
I appreciate your time.

cp2004, 2020-09-25:
Cooldown tracking is now *fully* functional on my development branch. I'm back to this plugin after the hectic run of support for [OctoPrint Upgrade to Python 3](https://community.octoprint.org/t/upgrade-your-octoprint-install-to-python-3/23973). Managed to crack this, and on a related note I can now implement #29 properly - it required an extensive re-write of the processing.

Implemented for 0.6.0

ssl-3, 2020-09-26:
Danke!

cp2004, 2020-10-29:
Hey @ssl-3 - could you test 0.6.0-rc.1 with this feature in it?

Can be installed by using the following URL in plugin manager > Get more > ...from URL:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

Please provide general feedback in the [feedback ticket](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/49) - an 'all is working fine' is valuable feedback since it lets me know people are testing this release. Thanks!

cp2004, 2020-11-01:
0.6.0 has been released

closed 2020-11-01 by cp2004