cp2004/OctoPrint-WS281x_LED_Status #17: [Feature Request]  New modes?  Custom modes (with some documentation)?

ssl-3, 2020-08-18:
**Is your feature request related to a problem? Please describe.**

It's not a problem, it's just something that would be nifty.

**Describe the solution you'd like**

I have a lot of ideas for new modes, most recently a "random" mode that sets each LED randomly on each cycle (as cycled by "delay," which I think should also be able to be non-integer).  I'd like to see adjustable modes that do this sort of thing.

But I'd also like a documented interface that can use (at runtime, ideally) that can can be used to implement custom modes.  In this way, the modes are easy to change and we (users) can cut-and-paste different modes without further involving the developer.

**Describe alternatives you've considered**

I've considered hacking up the source code and making random_mode a thing.  But that's not a long-term fix to producing other custom modes.

cp2004, 2020-08-18:
There is actually a random mode in 0.5 (Still on track to be released this week), which lights them all in random colours then changes one-by-one to a new (also random) colour. If it isn't what you want here, (Maybe you want them all off then a random LED is on?)

Now the thing with custom modes. There are a couple of specific design constraints to writing the modes, most notably the regular polling of the queue so the effect can end & start a new one - no blocking basically.
Custom code is not easy to run, it would be *difficult*, to say the least, as trying to write a way of interpreting user-defined effect code, into something that can be run in Python, would be hard. Not impossible, but hard. My other consideration is the number of people who would use this kind of feature. If I was to do it at all, it would be with 'upload your custom python script' and I would scan & try and import any functions, as effects. Interesting.

More modes and configuration options for those kind of modes is something I'm always looking at - In 0.5.0 the number of effects will nearly double (6 to 11), with a couple receiving 'tweaks'. I would also accept anyone who wanted to make their own effects - it's actually very easy to do, maybe it could be documented to encourage it. In many ways I would prefer that everyone benefitted from it, so contributing to the central plugin is preferred.

ssl-3, 2020-08-18:
Random colors (and intensities!  Randomize all the things!) is indeed the
thing that I'm after with the smaller request.

For the larger request:  Suppose that someone were to find an existing body
of work for WS281x LEDS that provided color-strings and would allow for
custom input. I will look for such a thing to see if it already exists and
report back if I find one that makes sense. (I'm new to the addressable LED
game [having been just introduced to it once I got your plugin to work] but
I think I understand the potential problems you outlined very well.)

I also admire your willingness to entertain the possibility of
evaluating/approving custom modes the old-fashioned way.  )   (My only fear
here is that you, personally, lose interest.  But I guess we can fork
things if that ever becomes important in the lifetimes of those who care --
just like anything else on Github.)

-ap

cp2004, 2020-08-25:
I have thought about this a bit, and have decided that I won't create a 'custom' effects feature, and instead will provide a detailed guide of how you could write an effect and get it into the plugin.

Reasons for this:
* Number of users who would use this feature:
   * Not many users will feel confident writing Python code to be used, so the amount of work vs. reward
* If it is done wrong (or without special care for some specific needs) it has potential to block & break the server, with how effects are being run in their current form
* importing the code into OctoPrint could cause other issues, including (potentially) security issues. I'm not sure exactly, but I get the feeling this will be tricky.
* Submitting PRs for effects here means everyone benefits, not just one person.

I'll create the doc at some point, the whole wiki is on the list to be re-written at the moment so it will go along with that.

cp2004, 2020-09-02:
Creating new modes is now documented in the new wiki (which I spent the weekend writing), specifically [Adding new effects](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/How-to-make-new-effects). 

There's a bunch of other information in the wiki too, if you think anything could be improved then let me know.

closed 2020-09-02 by cp2004

bccarlso, 2021-03-07:
> There is actually a random mode in 0.5 (Still on track to be released this week), which lights them all in random colours then changes one-by-one to a new (also random) colour. If it isn't what you want here, (Maybe you want them all off then a random LED is on?)

What action is supposed to trigger the "next random effect?" I finally got my LED strip working and this plugin is awesome! But I thought "random" would be different every time I turned the torch (flashlight!!) off then on. But it just refreshed the previous effect.

cp2004, 2021-03-07:
Random mode is not random effect, it just lights up the LEDs however it pleases. You can see an example of it in the documentation: https://cp2004.gitbook.io/ws281x-led-status/configuration/printing-effects#random

It picks a random colour, and sets a random LED to that colour. There's nothing else to it, there's none that cycles through the different effects. That would be an interesting idea though.

bccarlso, 2021-03-07:
Does reply here work? I noticed it was its own effect after I had posted
it, sorry about that! Maybe needs a different name :D But I think an actual
random effect each time you turn the torch on would be cool too.