cp2004/OctoPrint-WS281x_LED_Status #25: Getting random noise in the strip

stevenpaulr, 2020-08-25:
**What's your question?**

I have a 2812b strip, I got it to light up, but it keeps giving just random data to the LEDs. I've linked a video. This happens if it's printing or not.
[https://drive.google.com/file/d/1z_jEEHWWtezigX6MbDfpdDTJl2UEoM-g/view?usp=sharing](https://drive.google.com/file/d/1z_jEEHWWtezigX6MbDfpdDTJl2UEoM-g/view?usp=sharing)

cp2004, 2020-08-25:
Make sure you have a common ground between the power supply and the Pi - see in the image below, there is a ground connection going to both the Pi and power supply.
[image]

stevenpaulr, 2020-08-25:
That did it. Thanks!

cp2004, 2020-08-25:
Enjoy the plugin!

closed 2020-08-25 by cp2004

dtdor, 2020-12-03:
Having a crazy flickering/flashing problem when I choose anything except solid white. Grounds are common already so I'm a little stumped. I don't have a diode in place yet, haven't ever needed one before. I plan to try that and a 470 ohm resistor in the data line. Any suggestions would be great.