cp2004/OctoPrint-WS281x_LED_Status #26: LED stays white

jgrossrieder, 2020-08-27:
Hi,

I installed your plugin on octopi+raspberry 4.

I got the wizard, ran everything.
Also, I've been in the /boot/config.txt to set the `core_freq_min=500` as suggested in your doc https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/SPI-Setup-Running-without-root

Before setting the core_freq_min=500, I got some non-white lights, but it was cutting all the time.

Any idea what I could have messed up?

cp2004, 2020-08-28:
Hi there, @jgrossrieder 

Seems that the `core_freq_min` setting did do something, as the reason that is required is to prevent the cutting in & out because of dynamic core clock.

I'm wondering if there maybe is conflicting settings in the file, and it's now not doing what we want it to - could you upload/paste contents of `/boot/config.txt` here, so I can have a look? 
The other thing I am thinking, is that the wizard didn't detect the Pi model correctly, as setting `core_freq_min=500` should be in the wizard, if it detects Pi 4. Did you select to set `core_freq=250`? (Required for Pi 3 and earlier, not Pi 4, but would have shown up if it got detection wrong)

jgrossrieder, 2020-08-28:
Hi @cp2004 ! Thank you for the quick response.

I just went back to the file, and I got warning that a swap file is present. Anyway, here is the file I currently have(which still generate blank led):

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128
dtoverlay=w1-gpio                
```
I'm not sure what I can do to edit this file if it get overwritten by something else?

Regarding the core_freq, I tried multiple things, but currently it's not in the file.

cp2004, 2020-08-28:
The warning may be related to root ownership of the file - it does seem that the core_freq settings have not been saved. This should have been run & verified by the wizard, but maybe it is overwritten on a reboot

Try running the command directly, see what happens:
`sudo echo core_freq_min=500 >> /boot/config.txt`
Then you can verify that it is there by running `cat /boot/config.txt`, and the line should be there.
If it is there, reboot the Pi and you can check again. If not, then I will think about what to do from there.

Are you running OctoPi or a manual install of OctoPrint? I haven't seen this issue on OctoPi, hoping the RPi Foundation have not changed something on a recent firmware update.

jgrossrieder, 2020-08-28:
I ran the command, but I really think it's used by something else:

```
pi@octopi: $ sudo echo core_freq_min=500 >> /boot/config.txt
-bash: /boot/config.txt: Permission denied
```

I'm running OctoPi directly, freshly installed yesterday.

cp2004, 2020-08-28:
Permission denied.... I would have thought that using `sudo` sorted that, since it is a file owned by root. In linux there is no 'file is in use' like there is in windows, so I think it may be something different. Can you just confirm:
* Which Pi 4 model are you using  (1/2/4/8GB ram)
* Which OctoPi version you are using (0.17?)

I will have a think, and see if I can come up with something that means we can get into that file. I think that should be the only issue here, since if I set it up *without* core_freq_min defined, then my LEDs stay white too.

jgrossrieder, 2020-08-28:
Sure thing:

- Pi 4B 4GB
- OctoPrint 1.4.2 
- Python 2.7.16 
- OctoPi 0.17.0

I will try to shutdown the raspberry, and edit this parameters in windows tonight

jgrossrieder, 2020-08-30:
@cp2004 One additional weird thing: I unmounted my PI from its case, took the sd, and read it on my windows computer. 

The confix.txt file I got is:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README
core_freq_min=500

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128
dtoverlay=w1-gpio

```

so it contains the core_freq_min=500 and nothing else. I deleted the .swp file around.

cp2004, 2020-08-30:
Hmm. I can't reproduce this against my OctoPi install, the file is easily editable. Only difference is my `core_freq_min=500` is at the end, but that *shouldn't* make a difference as far as I understand it.

If you would like, you could run Adafruit's guide (uses their CircuitPython rather than `rpi_ws281x`) to rule out any hardware issues - It also runs with PWM instead of SPI like my plugin does, so it doesn't need the extra config. I'm not confident, but it may show we are not barking up the wrong tree.

Will continue to think if there is a way I/we can fix this.

jgrossrieder, 2020-08-30:
@cp2004 yes, sure I can, but I'm not sure to what guide you're referring? Do you have a reference?

cp2004, 2020-08-30:
Oops, I forgot to link it. Here:

https://learn.adafruit.com/neopixels-on-raspberry-pi

jgrossrieder, 2020-08-30:
> Oops, I forgot to link it. Here:
> 
> https://learn.adafruit.com/neopixels-on-raspberry-pi

It seems mandatory to use another pin than the GPIO 4. I've a 30 hour print in progress and the raspberry pi is below my structure, I cannot just change the connection now.

cp2004, 2020-08-30:
Yes that is true, I forgot that the PWM guide uses a different pin, apologies. Just wanted to rule out the hardware as an issue. Good luck with the 30 hours!

cp2004, 2020-10-07:
@jgrossrieder any update on this one? Just scrolling by and forgot where we got to trying to solve this.

jgrossrieder, 2020-10-08:
> @jgrossrieder any update on this one? Just scrolling by and forgot where we got to trying to solve this.

Sorry for the non-update, I receive another band led last week and I will try to replace saturday. I will also try to follow the adafruit guide to see the results.

jgrossrieder, 2020-10-10:
@cp2004 
I finally got the time to remove the whole printer(the rapsberry is screwed below the enclosure ).
So I just changed the DIN pin from the GPIO 10 to 18 and made a simple:

```
import board
import neopixel
pixels = neopixel.NeoPixel(board.D18, 105)
pixels.fill((0,255,0))
```

and it works, I've the first 105 led going to green.
Can I test something else?

Maybe I fucked up something in the config?
In your plugin, I've set:
```
Strip type: WS2812_STRIP
Max brightness 100%
Number of LEDs: 105
GPIO pin: 10
Frequency: 80000hz
DMA channel: 10
Invert pin output: unchecked
PWM channel: 0
```

Do you think I fucked up something?

cp2004, 2020-10-10:
So that's using Adafruit's circuit python library, which rules out the LED Strip as the problem I think. 

Maybe the next thing to try is with the `rpi_ws281x` library. I think at some point I would like to shift to the Adafruit libraries, but for now that is not possible due to a lack of support for Python 2 & a couple of other bits that make it a little *difficult*.

Try running this example file https://github.com/rpi-ws281x/rpi-ws281x-python/blob/master/examples/strandtest.py - not sure if the Adafruit guide gets you to install `rpi_ws281x`, so if not then `pip` or `pip3 install rpi_ws281x` will get you the library.

If *that* works, then you can edit this line in the file:

https://github.com/rpi-ws281x/rpi-ws281x-python/blob/4d3f84cea005093e22e0d48c46e19edd22a23a29/examples/strandtest.py#L14-L15

to change it from PWM to SPI (pin 10). If that works, then it is something in my plugin (which seems unlikely to me, since it sets the colours in the exact same way.

cp2004, 2020-10-16:
Hey @jgrossrieder,
How are things going for you with this. Got it working, or do we need to do more playing around?

jgrossrieder, 2020-10-17:
> Hey @jgrossrieder,
> How are things going for you with this. Got it working, or do we need to do more playing around?

I'm sorry, I took another plugin, less features than yours, but LED are working. Really sorry, but it's a pain to have to change pins for me.

closed 2020-10-17 by cp2004

kincaid-egotrip, 2020-11-11:
Hi. I am having the same issues as OP.
I have tried hooking the leds over to GPIO18 and running a pythonscript with the rpi_ws281x
rpi_ws281x works on PWM but not SPI. 
Using a Pi4 4GB. i guess the issue lies in spi and RPI4

cp2004, 2020-11-11:
@kincaid-egotrip To verify:
The LEDs work via GPIO18, using `sudo` to run
The LEDs **do not** work via GPIO10, not using sudo. 

I need to know if this is an issue with my plugin (I am not sure it is) or the SPI layer as a whole. It may also be an issue on newer revision Raspberry Pi's that needs to be patched upstream in the library. My RPI4 2GB and 4GB both work just fine with this plugin - maybe it is an issue in newer firmware, or Pi revision. Do you know what revision it is? (`cat /proc/cpuinfo` will be useful here)

The other thing I am yet to test is a newer Raspberry Pi OS image, to see whether that has any issues.

kincaid-egotrip, 2020-11-11:
GPIO18 sudo - working.
GPIO10 non-sudo - NOT working.

model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 162.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Linux octopi 5.4.72-v7l+

As a sidenote: I had issues adressing the GPIO pins. got an error: Oops - unable to determine board type... model: 17

GPIO version: 

gpio version: 2.52
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 02, Memory: 4096MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.

kincaid-egotrip, 2020-11-11:
It seems to me that the board revision might be related to mapping with rpi_ws281x SPI

Is there a way to make your plugin run with pwm?

cp2004, 2020-11-11:
> Is there a way to make your plugin run with pwm?

Yes - if you change the pin to 18, and run OctoPrint using `sudo`. Not recommended to have a webserver running as root and I have no idea what other issues you will have. It wasn't me who told you 😉 

I have the same board revision, but have not updated to the newer firmware (linux octopi 5.x) yet - let me try that now, see if that reproduces the issue.

kincaid-egotrip, 2020-11-11:
Its SPI clock scaling issues. Not sure if its easily solved in the spi lib. 
tried locking spi clock but no dice. The issue lies within the SPI lib. 
now, just figure out how i make octoprint run as root. (its behind a firewall, and on a seperate virtual domain so I am aware of the risk, and have precautions)

cp2004, 2020-11-11:
You have to supply the command line flag `--iknowwhatimdoing` to get past the protection *against* people doing it. Assume you mean setting `core_freq_min=500` in config.txt by 'locking the clock'

kincaid-egotrip, 2020-11-11:
Yes, tried core_freq=500 /250 in addition to core_freq_min=500/250 in different combos to try to make it play.
yes, i know what i am doing ;)