cp2004/OctoPrint-WS281x_LED_Status #27: [Feature Request] Time of day brightness

Gordontgopher, 2020-08-31:
**Is your feature request related to a problem? Please describe.**

No Problem
**Describe the solution you'd like**

There is a setting to turn off LEDs during the night, but a useful addition would be to allow setting the LED brightness by time, so that the LEDs could be set to be dim in the evening when there is typically less ambient light.
**Describe alternatives you've considered**

Set the brightness to a compromise setting!
**Additional context**

Hope it is self explanatory!

Thanks for a great plugin!

cp2004, 2020-08-31:
Added to 0.6.0 milestone. Thinking of two pairs of inputs, `min brightness` at `time` and `max brightness` at `time` too. Then just scale between the two. Will look into it once the existing stack is under control!

cp2004, 2020-09-28:
Pushed back to beyond 0.6.0 since it is more complex than I initially thought, and I have 0.6.0 planned to release by next week. Sorry, just haven't got the time to sink into this request at the moment. 🙁 
It will get done at some point...

Gordontgopher, 2020-09-28:
No rush :)