cp2004/OctoPrint-WS281x_LED_Status #29: [Bug] Heating display continues until progress is non zero

Gordontgopher, 2020-09-02:
**Description of the bug**

I've noticed that even when the print has started the 'heating' display is shown until there is enough progress to light the first 'progress' LED. In other words the display never shows a 'zero' progress state.
**Steps to reproduce**

1. Disable 'printing effect'
2. Enable 'printing progress'
3. Enable 'heating progress' for tool only
4. Start print

**What did you expect to happen**
Once print starts all LEDs would change from heating (red) to off then start to turn on green as print progresses.

**Logs and screenshots**

Can make these available if necessary
**Version of plugin and OctoPrint**

0.5.1
**(If applicable) Browser version & browser error console**

**Additional context**

cp2004, 2020-09-02:
Oh, yeah, I see the issue now. Print progress is only called when I receive progress from OctoPrint, which may not happen on 0%. Or, the first update happens during heating progress, so it is ignored.

Either way, will get this sorted for the next release. As it's not a large issue I won't be patching (unless bad bugs appear), only including in 0.6.0 once everything else is done.

Gordontgopher, 2020-09-02:
Oh yes, it's not a big issue by any standard, just slightly confusing at first :) Thanks for the quick response.

cp2004, 2020-09-27:
Fixed in `devel` branch, to be part of 0.6.0

cp2004, 2020-10-29:
Hey @Gordontgopher  - could you test 0.6.0-rc.1 with this feature in it?

Can be installed by using the following URL in plugin manager > Get more > ...from URL:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

Please provide general feedback in the [feedback ticket](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/49) - an 'all is working fine' is valuable feedback since it lets me know people are testing this release. Thanks!

Gordontgopher, 2020-10-30:
Ran a quick test, it seems that it goes from heating to first 'progress' LED lit straight away, even though the print is at 1%, I had expected a period of 'all off' before the first progress LED is lit, but maybe I had the wrong expectation :) ? It certainly fixes the original issue, thanks for looking at this.

commit 5091705a (2020-09-27): :bug: Make progress resume straight away when heating is finished

Closes #29
closed 2020-11-01 by cp2004