cp2004/OctoPrint-WS281x_LED_Status #31: [Bug]

Prutsium, 2020-09-15:
**Description of the bug**
When changing certain settings plugin stops working

**Steps to reproduce**
1: Go to Plugin settings and change idle effect to any other

**What did you expect to happen**
Well .... that it works :)

**Logs and screenshots**
Sep 15 08:54:55 raspberrypi octoprint[14628]:     effect_settings['delay'], self.max_brightness)
Sep 15 08:54:55 raspberrypi octoprint[14628]:   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_ws281x_led_status/effects/basic.py", line 201, in bouncy
Sep 15 08:54:55 raspberrypi octoprint[14628]:     strip.setPixelColorRGB(position[i], *color)
Sep 15 08:54:55 raspberrypi octoprint[14628]:   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 150, in setPixelColorRGB
Sep 15 08:54:55 raspberrypi octoprint[14628]:     self.setPixelColor(n, Color(red, green, blue, white))
Sep 15 08:54:55 raspberrypi octoprint[14628]:   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 143, in setPixelColor
Sep 15 08:54:55 raspberrypi octoprint[14628]:     self._led_data[n] = color
Sep 15 08:54:55 raspberrypi octoprint[14628]:   File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 54, in __setitem__
Sep 15 08:54:55 raspberrypi octoprint[14628]:     return ws.ws2811_led_set(self.channel, pos, value)
Sep 15 08:54:55 raspberrypi octoprint[14628]: TypeError: in method 'ws2811_led_set', argument 2 of type 'int'

**Version of plugin and OctoPrint**
Plugin: WS281x LED Status (0.5.1)
Octoprint: Version 1.4.2
Python: Python 2.7.16 

**Additional context**
This problem came in since 15-9-2020 and worked fine before.
Uninstalled plugin (and cleared data) but doesnt help

cp2004, 2020-09-15:
Hey @Prutsium!

Could you:
* Attach an `octoprint.log` so I can have a look at exactly what happens. If you can't find it, [there's a guide here](https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299?u=charlie_powell).
* Let me know what you have in the 'Strip Settings' Modal. Maybe you have changed one of the settings there and it is incompatible. 

I might add a lock function to prevent accidental changes of the strip settings since that can have quite bad consequences, and some error handling, to let you know when it goes wrong like this and the settings need to be changed.

Prutsium, 2020-09-15:
> <OctoPrint-WS281x_LED_Status@noreply.github.com>
> <mention@noreply.github.com>

What i do to replicate:
I change this to bouncy balls and it's gone (after restart octoprint)

Attached also the log and the error from the service:
Sep 15 17:26:58 raspberrypi octoprint[4923]:   File 
"/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_ws281x_led_status/effects/basic.py", 
line 201, in bouncy_
Sep 15 17:26:58 raspberrypi octoprint[4923]:     
strip.setPixelColorRGB(position[i], *color)
Sep 15 17:26:58 raspberrypi octoprint[4923]:   File 
"/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/rpi_ws281x/rpi_ws281x.py", 
line 150, in setPixelColorRGB
Sep 15 17:26:58 raspberrypi octoprint[4923]:     self.setPixelColor(n, 
Color(red, green, blue, white))
Sep 15 17:26:58 raspberrypi octoprint[4923]:   File 
"/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/rpi_ws281x/rpi_ws281x.py", 
line 143, in setPixelColor
Sep 15 17:26:58 raspberrypi octoprint[4923]:     self._led_data[n] = 
color
Sep 15 17:26:58 raspberrypi octoprint[4923]:   File 
"/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/rpi_ws281x/rpi_ws281x.py", 
line 54, in __setitem__
Sep 15 17:26:58 raspberrypi octoprint[4923]:     return 
ws.ws2811_led_set(self.channel, pos, value)
Sep 15 17:26:58 raspberrypi octoprint[4923]: TypeError: in method 
'ws2811_led_set', argument 2 of type 'int'
Sep 15 17:26:58 raspberrypi octoprint[4923]: 2020-09-15 17:26:58,520 - 
octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui 
_default) for {'query_string': 'l10n=
lines 1-18/18 (END)

Setting the effect back to default and plugin works again.

cp2004, 2020-09-15:
Can you upload the *full* file downloaded from the logging tab, so I can see the entire thing. Whole file, attached to the message in Github. Log excerpts are not good enough in this case unfortunately. If you cannot do this, I linked an explanation in my previous comment.

I think I *might* have an idea but its really hard to tell from just that excerpt, needs context.

Prutsium, 2020-09-15:
Wierd i added as attachment to the mail but seems the attacments then dont come over anyway here it is
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5226292/octoprint.log)

cp2004, 2020-10-18:
I spent such a long time on this one, it nearly didn't get solved. In the end, it needed 3 characters to (I think) solve it. I was never able to reproduce the actual issue, but managed to trace the reason for the error, and some poking later I found the problem. Solved, to be released with 0.6.0.

cp2004, 2020-10-29:
Hey @Prutsium  - could you test 0.6.0-rc.1 with this fix in it?

Can be installed by using the following URL in plugin manager > Get more > ...from URL:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

Please provide general feedback in the [feedback ticket](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/49) - an 'all is working fine' is valuable feedback since it lets me know people are testing this release. Thanks!

Prutsium, 2020-10-29:
Yes will do and report back

Thanks for taking actions

Sent: 10/29/2020 6:03:45 PM
balls effect, crashing runner (#31)

commit adbbdea4 (2020-10-18): :bug: Ensure the position of the LED is always an integer

Closes #31

3 letters. Hours of trying to track down the root of the error...
closed 2020-11-01 by cp2004