cp2004/OctoPrint-WS281x_LED_Status #32: [Feature Request]

cr0wm4n, 2020-09-21:
Is there anyway we can set the base temperature. for example,
my room with my printer is always around 22 degrees,
I have a strip of 20 leds and they always start half way along for progressing the temperature.
it woulf be good if we can set the 22 degrees as the start temperature and the Leds would then start from the 1st led.

cp2004, 2020-09-24:
On the milestone for 0.6.0, but I don't have a date for when that'll be released. Lots that needs to go into that release, that I am still working on, and I have had my hands full with [OctoPrint Upgrade To Python 3](https://octoprint.org/blog/2020/09/10/upgrade-to-py3/)

cr0wm4n, 2020-09-24:
Good to hear its coming :)
1 thing I am struggling with is the 1st pixel, I think its related to the ws281x library or SPI interface.
I am using gpio18 as i want to run as non root user, No matter what I do, on my 2812b strip the 1st pixel stays green, I even replaced it multiple times, i tried several buck converters to go from the main 12v psu to 5v and have ensure good common gnd between pi, strip and 5v side of the buck converters
Audio is disabled and blacklisted.
Even after I turn off this plugin, the first led stays green.

cp2004, 2020-09-24:
I knew I'd seen that before - this user mentioned it was the level shifter they were using, in https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/24

Maybe that will help?

cr0wm4n, 2020-09-24:
I already tried 2 llc's and without.
what i have found is if i move it to gpio18 from gpio10 andd run it as root then i do not have a problem. its only when its on spi gpio10 running as non root user

cp2004, 2020-09-24:
Does sound like it is an upstream `rpi_ws281x` issue, or at least with its SPI usage. 
I was unable to find anything conclusive about this elsewhere.... I *did* find a couple of reports of the 'first led is green' issue, but it was either look at the level shifter, *or* invert the signal - in this case using the little invert checkbox in plugin settings.

I haven't got any other ideas for this issue in that case. If I do end up with something, I will let you know.

cr0wm4n, 2020-09-24:
i agree i think it is upstream but thankyou for looking:)
i even searched everywhere to find a way to open up other gpio's to non root users, found nothing yet.

cp2004, 2020-09-24:
Leave it open! Still got the original request to implement...

> open up other gpio's to non root users
Due to the hardware of the Pi, and how has to interact with the LEDs, means that SPI is the *only* way it can happen, and the SPI pin on the Raspberry Pi is pin 10. Not sure there is a way around this, since it is hardware rather than software.

cr0wm4n, 2020-09-24:
there has to be a roundabout way, like a client server idea. server running as root controlling the gpio data and listening for a client packet stream or something like that

cp2004, 2020-09-24:
That would be a roundabout way, yes. I did also look into making a plugin that was compatible with WLED, so it was wireless (LEDs connected to ESP board) that would also potentially not have this problem, since LEDs could be run using proper PWM on the microcontroller. I know the Enclosure plugin kinda does the roundabout using command line scripts, this but in my opinion it is an awful implementation - and only allows for solid colours.

cr0wm4n, 2020-09-24:
i did try the enclosure plugin to be honest, and rgb status.
neither of them i was happy with.
this is the only one that does what i wanted out of the box sort of thing..
the base temp thing is the 1 big improvement i am looking for. as for the green led, if i have to i can live with that :)

cr0wm4n, 2020-09-26:
i think i found a 281x server that can be utilised from a webserver :)
i will have a play with it later
https://github.com/tom-2015/rpi-ws2812-server

cp2004, 2020-09-27:
Feature implemented & ready for version 0.6.0

cr0wm4n, 2020-09-27:
That was quick ;))
I will install devel and try it after this current print finishes :)

cp2004, 2020-09-27:
Be warned of bugs.... lol

It should be ok but I tend to break quite a things and then fix them later in final testing.

cr0wm4n, 2020-09-27:
ok, installed and the feature appears to be working, but there may be a bug, i will test more.
what happened was i aborted a print a few times, when i aborted it  all the leds went white instead of like the rainbow effect i had in place at the end of the print. 
this didnt happen every time, just once or twice but i will test more to see if i can find any pattern to it.

cp2004, 2020-10-29:
Hey @cr0wm4n  - could you test 0.6.0-rc.1 with this feature in it?

Can be installed by using the following URL in plugin manager > Get more > ...from URL:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

Please provide general feedback in the [feedback ticket](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/49) - an 'all is working fine' is valuable feedback since it lets me know people are testing this release. Thanks!

commit 9f6dd198 (2020-09-27): :sparkles: Allow setting baseline temperature

Closes #32
closed 2020-11-01 by cp2004