cp2004/OctoPrint-WS281x_LED_Status #34: [Bug]LED SK6812 Strip showing same wrong color but passing all tests

Lisheng2016, 2020-09-24:
**Description of the bug**
No matter how i change the led pattern,the strip always show a ranbow color like in the picture.
But the tests of all green,blue,white all are showing correct results.I'm out of ideas could you show me some ways to debug?I would be very happy to help thanks!

**Logs and screenshots**
[image]

**Version of plugin and OctoPrint**
plugin: latest
Octoprint: version 1.4.2 on Python 2.7.16 + OctoPi 0.17.0

cp2004, 2020-09-24:
When you say
> No matter how i change the led pattern
Where are you changing it? If there is nothing happening on the printer, it won't change from either the 'startup' mode or the 'idle' mode. You also have to save the settings for them to be passed to the strip.

If you think you *are* changing the right settings, could you upload an `octoprint.log` and I will take a look when I get a chance later.

Lisheng2016, 2020-09-24:
> When you say
>
>
> If you think you _are_ changing the right settings, could you upload an `octoprint.log` and I will take a look when I get a chance later.

Thanks for the reply. I've downloaded the rpi_ws281X library and tested the strip,no problem when running from command line.
And yes, i saved everytime i changed the pattern of idle mode.
Now i'm controlling direct from the pi,as i only want some lighting.

closed 2020-09-24 by Lisheng2016

cp2004, 2020-09-24:
So this issue is... not resolved? 🤔 

If you would like me to look into it further, I would need some more information - if you want to:
* Enable debugging logging from the bottom of the 'Features' tab, then save settings
* Try and get the plugin to work (change settings, connect the printer, start a print - something you would expect to have an effect)
* Send me the `plugin_ws281x_led_status_debug.log` file, from OctoPrint's logging panel.

It could be, you have come across a bug, but I cannot tell without any kind of log. And if it is a bug, I like fixing them...

Installing RGB Status doesn't get the problem fixed (maybe it affects someone else?), and to be honest there's no activity on that plugin, or any support so you are better off with me 🙂

Lisheng2016, 2020-09-26:
> So this issue is... not resolved? 🤔
>
> If you would like me to look into it further, I would need some more information - if you want to:
>
>     * Enable debugging logging from the bottom of the [...]

I'll post the logs as soon as possible.And i was actually wondering if i used the plugin the right way before posting the issue.

Lisheng2016, 2020-12-07:
Just migrated to python3 and everything works like a charm.Great Plugin,happy playing with it now.