cp2004/OctoPrint-WS281x_LED_Status #35: Add an option to make the Torch Mode button a toggle

JohnTRoth, 2020-09-28:
There are times when I'd just like the printer / print bed to be highly visible. Torch mode is perfect for that, but it's restricted to the timed torch mode settings that are pre-programmed. Personally, I'd prefer it to be a toggle button. I suspect this could be accomplished with minimal effort by passing in a torch duration of 0 or -1, though you may want to go all-out and add an independant toggle button on the interface, or perhaps a checkbox for it in the options screen.

Thanks!

cp2004, 2020-10-18:
Implemented through https://github.com/cp2004/OctoPrint-WS281x_LED_Status/commit/62f0ee5e012a49207a413ce2094c03c58822cf11 for 0.6.0

closed 2020-10-18 by JohnTRoth

roboraptor, 2021-01-06:
Would it be possible if you set Torch to toggle, to have the swipeswitch button next to it? (same like the lights on/off)

cp2004, 2021-01-06:
Open another request, definitely possible, but I would forget on a closed issue.