cp2004/OctoPrint-WS281x_LED_Status #36: [Bug]: Interference with Dashboard plugin when re-starting Octoprint

fnsign, 2020-10-06:
**Description of the bug**
There is a new version of the [Dashboard plugin ](https://github.com/j7126/OctoPrint-Dashboard)available in version 1.15.0 (from a new maintainer of the plugin). This new version interferes with the WS281x_status plugin, when you want or need to restart Octoprint (e.g. when installing a new plugin). I deactivated all installed plugins and re-activated them 1-by-1 to see which plugin causes the issues with Dashbaord. And it was WS281x. As I can't say which plugin is the source of the issue I opened a bug report there, too [#214](https://github.com/j7126/OctoPrint-Dashboard/issues/214).
Something must have changed, since v1.13.0 is causing no issues.

**Steps to reproduce**

1. Both plugins must be installed and activated
2. Go to 'Restart Octoprint' and confirm the prompt.

**What did you expect to happen**
The expected behaviour is that Octoprint restarts, showing the prompt for "Attempt to reconnect" and after that "Reload".
Unfortunatelly the service is not restarted correctly. The GUI is still working, but no inputs are taken.

**Logs and screenshots**
see  [#214](https://github.com/j7126/OctoPrint-Dashboard/issues/214)

**Version of plugin and OctoPrint**
Dashboard: 1.15.0
WS281x: 0.5.1
Octopi: 0.17
Otcoprint: 1.4.2

**(If applicable) Browser version & browser error console**
MacOS Mojave 10.15.6
Safari 14.0

Thanks in advance!

cp2004, 2020-10-06:
In contact on the other issue, it is (kind of) within the dashboard plugin, so it is their problem. Might send in a PR to get it fixed.

fnsign, 2020-10-06:
Maaaan, you are fantastic!! Thank you so much for taking care so quickly and finding the problem!

cp2004, 2020-10-06:
Going to close this, since my PR has been merged. when they cut a new release, it will be solved.

closed 2020-10-06 by cp2004