cp2004/OctoPrint-WS281x_LED_Status #37: Change LED Direction

wkoetz, 2020-10-15:
Is it possible to change the LED direction?

cp2004, 2020-10-15:
Not currently. @11chrisadams11 was working on that in #14 , and I had plans to turn it into a global setting, reverse direction.

cp2004, 2020-10-25:
From 0.6.0 it will be possible to reverse the order of progress effects. @wkoetz is that what you wanted? I don't have plans to convert any other effects before the next release.

wkoetz, 2020-10-25:
Great and yes, this is exactly my wish! THX and take care, stay healthyWilfriedVon meinem Samsung Galaxy Smartphone gesendet.

cp2004, 2020-10-29:
Hey @wkoetz  - could you test 0.6.0-rc.1 with this feature in it?

Can be installed by using the following URL in plugin manager > Get more > ...from URL:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

Please provide general feedback in the [feedback ticket](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/49) - an 'all is working fine' is valuable feedback since it lets me know people are testing this release. Thanks!

wkoetz, 2020-10-31:
Yes, now it works great for the progress bar. My first thinking was it doesn't work, because it has no effect to "Color Wipe" at startup effect. Maybe you can add a reverse Color Wipe as well?
Nevertheless great work and I am happy with that!

cp2004, 2020-10-31:
Maybe it will be for all effects in the future, I will think about it for future releases.

cp2004, 2020-11-01:
0.6.0 has been released.

closed 2020-11-01 by cp2004