cp2004/OctoPrint-WS281x_LED_Status #38: Strobing Effects?

aleroy, 2020-10-16:
**Should the effects be transitioning smoothly between colors?**

I either have something setup incorrectly or this plugin may need to come with a photosensitivity warning. I'm hoping the former.

I have 93 LEDs hooked up via a level shifter to my pi and a 330 ohm resistor before the 1st LED. It's powered by a 5amp buck converter from the printer's supply. My Pi runs on a separate supply. The grounds are connected via the level shifter (simple BSS138 transistor style). As for settings: I set the LED count, dropped the overall brightness to 25%, and left everything else default.

The progress status seems to work as expected. Temperature status sets all the LEDs blue and then progressively sets each LED first to violet then to red. Printing progress is the same but it goes OFF to RED to GREEN. 

But I expected the printer effects would transition smoothly---like the strandtest for Adafruit's Neopixel library on Arduino. I get a lot of blinking instead. For example, the Rainbow effect, even with the delay set to 0, has all the LEDs strobing magenta, then cyan, then yellow, and repeat. Is this how it's supposed to work?

aleroy, 2020-10-16:
[Video Here](https://drive.google.com/file/d/1W1WvoYu0RET-zpPKbFB66chO5lhykrAB/view?usp=sharing)
The frame rate is a bit of an issue for capturing the blinking. You can at least see it's sticking to the same 3 colors.

Resource Monitor shows 6% - 10% CPU.

I also added an extra ground wire from a Pi ground pin to the strip ground and tried a different level shifter. Neither helped. So then I tested the stip. Connected to an Arduino running the Neopixel "strandtest" it worked perfect.

One other thing I noticed. The 1st LED is a slightly different color than the rest of the LEDs on the rainbow effect. On the bounce effects it's always green. Connected via Arduino it was fine.  GPIO 10 on the Pi 3B+ is also the MOSI pin. Could that be the problem? I changed it to GPIO 5, but now my octoprint won't startup.

aleroy, 2020-10-16:
Octoprint is starting now, but the strip doesn't work on GPIO 5 ... or 16.

cp2004, 2020-10-16:
Looks like you just need to turn the delay value up - try something like 50ms for that effect. It's just cycling *really* fast through the effect, it appears.

cp2004, 2020-10-16:
Oh and WRT the pin numbers, only 10 will work. Per the documentation, and I suppose it could be clearer *why* that is the case, but there's a lot of information there.

Direct link: https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/LED-Strip-configuration#gpio-pin

aleroy, 2020-10-16:
That just makes it blink slower.

cp2004, 2020-10-17:
Something's definitely up here. I can't tell if it is a bug with the effect itself (quite possible, with certain combination of settings or something).

I might be able to look into it at some point, although in the meantime would you be able to enable 'debug logging' (under features) and then upload the `plugin_ws281x_led_status_debug.log` file here. I'm not confident that it will show anything, but what it *might show* if there is any strange data coming through to the process, causing a refresh of the effect.

In the next release, I am planning to add some more useful logging, to help with diagnosing issues like this. I haven't been able to reproduce this, and as a side note the effects should all look like the visualisations [in the documentation](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Effects) so you can see when it is going right. I will do some code analysis to see if I can spot something that *might* cause this.

aleroy, 2020-10-17:
Take your time. Solid colors get the job done and still look pretty cool. Thanks for your timely responses and more-so for the plugin and documentation. I appreciate the effort you've invested here.

I'll get you some log data once my print finishes.

KilianB, 2020-10-18:
A strobing effect can appear if your psu isn't able to adjust to the voltage spikes quickly enough. My setup only started working smoothly once I added a bigger capacitor just before the led across the power line. Do you have anything at hand to give it a try? Otherwise do you have a voltmeter to get some readings?

aleroy, 2020-10-18:
@KilianB I have plenty of power and plenty of capacitance. Just to be sure, I hooked it up to an 8amp buck converter (which is connected to a 350W 24V Meanwell PSU).

@cp2004 It's the number of LEDs. I plugged in a simple 8 LED light ring and worked perfect. It's just taking too long to update 85 LEDs. I don't know why my Arduinos can update all of them and the Pi (octoprint) can't.

Also: attached the log
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5398314/plugin_ws281x_led_status_debug.log)
.

cp2004, 2020-10-19:
Would you be able to plug in your larger strip, and keep the settings for 8 LEDs, and see if that works? If it does, it might be good to try and find the point it starts to fail. I have a strip of 50 here, and it is able to keep up with that number.

aleroy, 2020-10-20:
I made the ring myself and didn't solder in a data out connection. The LEDs I have installed were cut down to various sizes to fit the lengths where they're installed. So I can connect each of them separately. I tested strips of 8, 14, 20, & 33 LEDs. It started to fail at 20. With the strip of 20 LEDs connected I reduced the number of LEDs specified in the settings. I turned the strip off using the on/off button in octoprint between setting changes (otherwise the latter LEDs wouldn't get cleared). When I got to 15 it something interesting. It looked to be working as expected ... except there were 16 LEDs cycling smoothly and the 1st LED in the strip was solid blue.

I noted the current and voltage readings for each test. The 20 LED strip set at 20 LEDs peaks around 0.25 A. Set at 15 it uses 0.07 A. I have the max brightness setting at 25%.

Also, when I rebooted the Pi and cycled the strip's power, the 1st LED would cycle with the rest, but there were still 16 LEDs going instead of the specified 15. 

I repeated these tests with the 33 LED strip with the same results.

Then I got curious about my 25% max brightness setting. I bumped it up to 50% and found I was able set the number of LEDs up to 17 before the blinking started. It maxed out there, at 100% max brightness only a setting of 17 would cycle smoothly (with 18 actually changing).

cp2004, 2020-10-21:
Sorry I didn't get back to this sooner. I can't see the two options from the source, they are all different for me - only one WS2812 option. Could be a visual bug with the select I guess.

I'm going to admit I have limited what the problem could be here. I don't think it is processing related, since I can up the number of LEDs to 100s and it still copes just fine, and I have reports of people using 200 LEDs. You seem confident it is not power/general wiring related, and say that the strip works fine with Arduino & the examples. Now it *could be*, that the LEDs do not like being driven over SPI as opposed to PWM on the arduino. I don't know *why* this would be, but it is possible.

Would you mind checking the list of config from [SPI Setup](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/SPI-Setup) - the only thing I can think of that may affect it is the SPI buffer size, since that means you can drive more than a handful LEDs well.

Hopefully you can make sense of my thoughts above... any more questions let me know 🙂

aleroy, 2020-10-23:
Yeah, no go.

I haven't tried the rpi_ws281x examples yet. That should have been one my first steps.

cp2004, 2020-10-24:
I'm not sure what else to suggest, since this is not something I've seen before... Running rpi_ws281x examples (use pin 10, no root) should narrow it down to what is causing the problems.

aleroy, 2020-11-02:
I'm going to close this issue for now. It seems obvious that the issue is outside the scope of your plugin. I'll continue trying to figure this out, but it might be awhile as it's not a big priority for me. I'll be sure to update you if/when I figure it out.

closed 2020-11-02 by aleroy

cp2004, 2020-11-02:
Its outside the scope of my knowledge of these LEDs as well, I've never seen things like this before. Hopefully you do figure it out, since LEDs are cool 🙂 
Good luck.