cp2004/OctoPrint-WS281x_LED_Status #39: [Bug] torch effect does not revert to previous effect

KilianB, 2020-10-17:
**Description of the bug**

When clicking the torch mode button the strip starts the effect specified effect but does not revert to the previous effect even after the time has passed.
 
**Steps to reproduce**

Click on the torch icon.

[image]

**What did you expect to happen**

Torch effect to play for 5 seconds
Return to idle effect afterwards

**Logs and screenshots**

**Version of plugin and OctoPrint**

WS281x LED Status (0.5.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_ws281x_led_status

 os:
|      id: linux
|      platform: linux2
|  plugins:
|      pi_support:
|          model: Raspberry Pi 4 Model B Rev 1.1
|          octopi_version: 0.17.0

**(If applicable) Browser version & browser error console**

Chrome: Version 86.0.4240.75 (Official Build) (x86_64)

**Additional context**
- The test modes to test individual leds are not working either. 
- Changing the idle effect to another effect in the settings panel does not change the effect until restart?

KilianB, 2020-10-17:
I just looked at the code and saw that for the test case `$('#led-test-red').bind('click', function() {send_m150(255, 0, 0)})` an m command is send. Does a printer have to be connected in order for certain functionalities to work?
So far I have only set up octoprint and the led strip. A printer is yet to be connected. I'll do a test with the virtual printer tomorrow to see if some of the bugs disappear.

cp2004, 2020-10-18:
The LED test buttons do need the printer connected. I hadn't really thought about that, clearly. At the time of developing, that was the fastest way to get the function across, since M150 was already a thing. Shouldn't be too difficult to add an API command instead of gcode.

It looks like the root of the problem could be, as you say, that the printer is not connected. So you don't get any of the other effects to run. Since in the debug log, there is no effect sent that is not `torch`, so if I remember right the default should be `startup` (so *maybe* it goes back to that?). Need to test what happens here, if printer is not connected. I also know I re-wrote a lot of the saving of previous states, to help implement #16 & #29 in my development branch. So *some* of this might be a bit better when I can release that.

Mind if I change this, from less of a bug report to an improvement, something like `Improve experience when printer is not connected`?

KilianB, 2020-10-18:
Sure, go ahead. I can confirm that after connecting the virtual printer the functionality is present. Do you accept pull requests? I might take a look at it while I am waiting for my camera module.

cp2004, 2020-10-18:
Go for it! Be sure to look at the [`devel`](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/tree/devel) branch for the current development. API commands are defined (with parameters etc. etc. ) here:

https://github.com/cp2004/OctoPrint-WS281x_LED_Status/blob/814e51ffe1600fdadeac1b8dbacf6b4a59580ce1/octoprint_ws281x_led_status/__init__.py#L231-L235

And then this is how you can call the M150 handler internally (currently only way to pass custom solid colour, rest is settings based):
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/blob/814e51ffe1600fdadeac1b8dbacf6b4a59580ce1/octoprint_ws281x_led_status/__init__.py#L510-L511

have fun!

cp2004, 2020-10-25:
@KilianB - are you still looking into this? I would like to push a release by the end of the week (there's quite a lot of new things that need to be released!). No worries if you no longer have time, I will do it so it can be included in 0.6.0.

KilianB, 2020-10-25:
Yes I'll be looking into this. Are we talking within the next 6 days or today? The _start and end of the week_ are always a bit ambiguous depending on where you are ;)

cp2004, 2020-10-25:
Ah, no 6 days. I'm trying to add some of the bigger new features in while I have time today, then throughout the week will be testing & polishing them. So by next Sunday, 1st Nov. 🙂

cp2004, 2021-01-18:
Implemented via the above commit for 0.7.0

cp2004, 2021-03-05:
0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

commit 6105cd75 (2021-01-18): :sparkles: Improve utilities with KO bindings

Closes #39
closed 2021-03-11 by cp2004