cp2004/OctoPrint-WS281x_LED_Status #44: [Bug] LEDS stay in paused mode on filament runout

cr0wm4n, 2020-10-19:
I currently have a bad wire on my filament runout sensor (I will fix it after this current print), this is causing the filament runout to trigger and the printer goes into a paused state.
When I resume the print the LEDS stay in a paused state for what seems an excessively long period. 
I have to turn on torch mode to be able to see the print. Is there a way to have the LEDS resume when the print resumes ??

cp2004, 2020-10-19:
I *think* the LEDs should stay in pause mode until they get another 'event' to change state. In this case, I think it might be until the next % on the progress bar. As part of #29 I re-wrote parts of this, so it can remember what was there before. Should be possible to easily react to a PrintResumed event, and put back the old progress bar.

cp2004, 2020-10-21:
Ready for next release (0.6.0)

cr0wm4n, 2020-10-22:
:)
i will update again thankyou :)

cp2004, 2020-10-29:
Hey @cr0wm4n  - could you test 0.6.0-rc.1 with this feature in it?

Can be installed by using the following URL in plugin manager > Get more > ...from URL:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

Please provide general feedback in the [feedback ticket](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/49) - an 'all is working fine' is valuable feedback since it lets me know people are testing this release. Thanks!

cr0wm4n, 2020-11-01:
yes of course i will,
i will install when my current printjob is complete :)

commit 451dd8e2 (2020-10-21): :sparkles: Explicitly put back progress effect on print resume

Closes #44
closed 2020-11-01 by cp2004