cp2004/OctoPrint-WS281x_LED_Status #49: [RC Feedback] 0.6.0rc1 General Feedback

cp2004, 2020-10-29:
Please provide **general** feedback in this ticket, for 0.6.0rc1. An 'all is working fine' is valuable feedback too, as it lets me know that people are in fact testing this release.

### Installation

Install this release by using the following URL in the plugin manager > Get More > ... from URL

```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/0.6.0-rc.1.zip
```

## Todo for stable release
* [ ] #39 - hopefully by contribution shortly

## Identified bugs
None so far 🙂 

**❤️ Thank you for your contribution in helping test this release!**

Prutsium, 2020-10-30:
All works ... no crashes when changing effects (also Bouncy Balls)

cp2004, 2020-10-30:
@Prutsium Thank you for testing it!

Gordontgopher, 2020-10-31:
Ran a quick test regarding #29 it seems that it goes from heating to first 'progress' LED lit straight away, even though the print is only at 1%, I had expected a period of 'all off' before the first progress LED is lit, but maybe I had the wrong expectation :) ? It certainly fixes the original issue.

cp2004, 2020-10-31:
How many LEDs do you have? If the print is at 1%, the LED will be lit but quite dim - 0% *should* be off. As progress increases, the led will brighten until it moves to the next one. Best demonstrated with two different colours, as it blends between the two as progress increases. Thank you for testing it, I'll aim to fix the issue in #50 and then look to release the end of tomorrow I think.

Gordontgopher, 2020-10-31:
I've got 18 LEDs, so I wasn't expecting the first LED to be lit until ~5%. I do have the LEDs set to quite low power, so it may be I'm not seeing as much difference between dim and bright. It's certainly fixed the issue of still showing heating at the start of the print.

JohnTRoth, 2020-11-01:
It's all working perfectly for me - thanks!

cp2004, 2020-11-01:
0.6.0 has been released.

closed 2020-11-01 by cp2004