cp2004/OctoPrint-WS281x_LED_Status #50: WS2812 Plugin not showing up in Pi Zero installation

Prutsium, 2020-10-31:
Pi Zero with Original Octoprint image

**What did you expect to happen**
To see the plugin

**Logs and screenshots**
2020-10-31 00:42:36,441 - octoprint.server.views - ERROR - Error while retrieving template data for plugin ws281x_led_status, ignoring it
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/views.py", line 684, in fetch_template_data
    wizard_required = implementation.is_wizard_required()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/__init__.py", line 238, in is_wizard_required
    for item in self.get_wizard_details().values():
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/__init__.py", line 244, in get_wizard_details
    return wizard.get_wizard_info(self.PI_MODEL)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/wizard.py", line 74, in get_wizard_info
    "core_freq_set": is_core_freq_set(pi_model),
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/wizard.py", line 101, in is_core_freq_set
    if int(pi_model) == 4:  # Pi 4's default is 500, which is compatible with SPI.
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5468274/plugin_ws281x_led_status_debug.log)
[octoprint(1).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5468275/octoprint.1.log)
[image]
[image]

-->

**Version of plugin and OctoPrint**
 Using pip of "/home/pi/oprint/bin/python3", Version 20.2.4
WS281x LED Status (0.6.0rc1)

cp2004, 2020-10-31:
Thank you for reporting this! This is why I like release candidates... Something in my wizard rewrite has gone wrong I guess.

cp2004, 2020-10-31:
@Prutsium Could you share the output of `cat /proc/device-tree/model` please? I think this might be that the Pi Zero is reporting a non-number (instead of Pi 4, Pi Zero) and is messing with things. Thanks!

Prutsium, 2020-10-31:
pi@octopi:~ $ cat /proc/device-tree/model
Raspberry Pi Zero W Rev 1.1

cp2004, 2020-10-31:
Fixed ready for stable release of 0.6.0

commit 1d8d45b4 (2020-10-31): :bug: Fix detection of Raspberry Pi Zero

Closes #50
closed 2020-11-01 by cp2004