cp2004/OctoPrint-WS281x_LED_Status #52: [Feedback] 0.6.0 Feedback ticket

cp2004, 2020-11-01:
Ticket for you to provide *general* feedback on the release. Maybe you like a particular feature, or have a quick question about something new in this release. Or anything else!

For feature requests and bug reports, please [open a new issue](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/new/choose) using the templates.

Thank you for using this plugin. If you have enjoyed using this, please consider [supporting it's development!](https://github.com/sponsors/cp2004)

cr0wm4n, 2020-11-01:
ahh 0.60 already released, im upgrading to that now

Prutsium, 2020-11-01:
Installed 0.6.0 on a Zero and "Yes" it works.
Thanks for the quick fixing.

cr0wm4n, 2020-11-01:
yes the temp thing works fine 👍

cp2004, 2020-11-01:
Thank you both for the feedback 🙂

cr0wm4n, 2020-11-01:
thankyou for completing this request :))

Drew3164, 2020-11-02:
Not working? Did the update and the whole plugin disappear tried reinstalling the plugin and does not appear in the setting/ plugins?

cp2004, 2020-11-02:
@Drew3164 Please open a full bug report, I suspect the answer lies in the logs.

cp2004, 2020-11-10:
@Drew3164 - just checking in, guess this is no longer an issue?

Drew3164, 2020-11-10:
No I fix it by reinstalling a new octoprint and your new update help out
alot.

Thanks for your help and getting back to me..

Keep up the good work.

closed 2021-02-13 by cp2004