cp2004/OctoPrint-WS281x_LED_Status #53: [Bug] Minor bug, notification on install

cr0wm4n, 2020-11-07:
when i installed this and i needed to restart, the notification told me ws2812b was successfully installed

cr0wm4n, 2020-11-07:
to recreate... 
uninstall marlin eeprom editor
complete the octoprint restart as requested
reinstall marlin eeprom editor
complete the octoprint restart as requested
your announcement opens up
click finish
then you get this notification

cp2004, 2020-11-07:
Yeah I fully get it now, and can reproduce it. Just got to figure out who's problem the bug is 🙂 

Would probably happen with *any* plugin that uses a wizard. Seems to be the `onBeforeWizardFinsh` that should not be called unless the WS281x wizard is shown.
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/blob/b06d37becf7a146f877e07c1ed81ae4433ab9b70/octoprint_ws281x_led_status/static/js/ws281x_led_status.js#L96

cr0wm4n, 2020-11-07:
glad its not just me now lol, its hardly much of a bug as its only a notification :)

bouba830, 2020-11-19:
Hello, every 2 to 3 minutes I have a blink, do you know why?
material: WS2811 12V

cp2004, 2021-01-18:
Fix implemented via the above commit for 0.7.0

cp2004, 2021-03-05:
Hey @cr0wm4n, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

commit f1a64b11 (2021-01-18): :bug: Do not trigger notificaton if wizard is not loaded

Closes #53
closed 2021-03-11 by cp2004