cp2004/OctoPrint-WS281x_LED_Status #54: [Feature Request] - Support for Multiple instances of octoprint on one raspberry

nobentrix, 2020-11-08:
Dear cp2004,

Would love to have support for multiple instances of octoprint on one raspberry :)

- I use 4 printers and it would be wounderfull if every printer has his own "LED Status Bar" which can be coltrolled over seperated output pins from one raspberry.

I try to set up the GPIO pins (11,9,8,7) in the "LED Strip configuration" window for each instance, but you write that this not work actually.

>>Technically possible to use other pins, but as OctoPrint can't (by default) be run as root it cannot be changed<<

So if the pinout can not be changed is it possible to set "sections" in one long LED Stripe?

e.g.

- one stripe for 4 pritner with 192 leds in total. 48 LEDs for each Printer

- Printer 1. LED data from 1-48
- Printer 2. LED data from 49-96
- Printer 3. LED data from 97-144
- Printer 4. LED data from 145-192

So that in the "LED Strip configuration" can set up the START and END point for each instances of octoprint.

Sorry for my bad english I used the google translator.

Thank you for your support.

KR Oliver R.

cp2004, 2020-11-08:
I've been asked about this before, and I have to say no. Why?:
* The LEDs require very precise timing, and as a result the SPI interface can only be used by one process (ie, one OctoPrint instance)
* This is a limitation of the underlying library `rpi_ws281x` - it does not allow more than one instance to run on the SPI pin, which your solution would require.
* As a result, the only way to do this would be a separate app that receives data, from multiple OctoPrint installs. I haven't got time or resources to develop such a complex solution
* Also, I'm developing another solution, using [WLED](https://github.com/AirCookie/WLED), which would allow for this. It is wireless, so is not affected by the limitations of the Raspberry Pi. Long way off, but it is in the workings.

Hopefully that is clear. Its just not possible with the hardware and software that drives the LEDs. Sorry!

closed 2020-11-08 by cp2004

nobentrix, 2020-11-08:
Dear cp2004,

no reason to say sorry! I am super happy to get such a helpfull and clear answer.

Thank you for your nice support.

KR Oliver R.