cp2004/OctoPrint-WS281x_LED_Status #55: Motion sensor with WS281x Plugin?

dtdor, 2020-11-11:
Currently I've been controlling my 2812 LEDs with an arduino on my printer. The lights and arduino are powered by the same 5v power supply that I used to recently set up octoprint (supplies up to 10 amps). I have a PIR motion detector connected to the arduino to turn on the lights when I walk into the kitchen at night and it works well but it's pretty basic. I think this plugin sounds awesome and clearly adds a lot more function to the LEDs but I also don't want to lose the motion detector. Is there a reasonably simple way I can use both? I'm pretty bad with coding except for arduino (where I'm basic but functional).

Thanks!

cp2004, 2020-11-11:
This is a good idea, makes quite a lot of sense. Motion detection is not a feature I would want to implement myself into the core of the plugin, since it makes it quite complicated where it could be easily solved externally. Some possible solutions for you:
* Wire a mosfet or something similar along the power wire to LEDs, to easily be able to turn them on/off using the arduino (probably the least extra work). If I had a mosfet, I might do this. I also have an ESP8266 board, so:

* If you had something like an ESP board, make an API call to `/api/plugin/ws281x_led_status` along the lines of:
   ```http
   POST /api/plugin/ws281x_led_status HTTP/1.1
   Host: example.com
   Content-Type: application/json
   X-Api-Key: abcdef...

   {
     "command": "toggle_lights",
   }
   ```
   Note that the API stuff is undocumented, so it may change without notice. Or that could become a request in the future, a stable API...

I think external is the way to go with this, I did have a request to use a GPIO button for on/off, but don't want to put it into the plugin myself. Maybe I *will* think about a more stable way of external scripts interfacing, to cover *whatever* people may want to add themselves.

dtdor, 2020-11-11:
That’s a great idea, thanks. I think I will wire in the mosfet. I would be
interested in a feature that adds this into the plug-in so that the lights
can be animated somewhat to turn on gradually (2-3 seconds) to help
nighttime eyes adjust a tad. But maybe I can do that in the arduino code
with the mosfet by gradually increasing voltage from the forward voltage up
to 5v to get a smoother “on.”

I’ll let you know how it goes.

Thanks!

cp2004, 2020-11-11:
> I would be
interested in a feature that adds this into the plug-in so that the lights
can be animated somewhat to turn on gradually
Interesting... I could agree, this would be cool.

cp2004, 2020-11-29:
Somehow forgot to close this issue, since you got it worked out. Will implement fading in and out at some point, but since its not related to this issue I can close.

closed 2020-11-29 by cp2004

dtdor, 2020-11-29:
Sounds good.