cp2004/OctoPrint-WS281x_LED_Status #56: Blink

bouba830, 2020-11-19:
**What's your question?**
Hello, every 2 to 3 minutes I have a blink, do you know why?
material: WS2811 12V

cp2004, 2020-11-19:
No, I don't know why. You're going to have to give me more details.

I have seen some occasional blinks, which I think are down to the CPU swapping processes about (unconfirmed, just a theory), but if it is worse then maybe you need to check all the connections.

bouba830, 2020-11-20:
I don't know what to tell you, details and well I have a WS2811 led strip in 12V of 60 leds on a raspberry PI 4 B +, a 12V power supply connected to the + and the - and A cable to the GPIO 10 and another on the - of the RASP, when I run a print the programmed colors work correctly throughout the process but from time to time at irregular intervals I have a blinking, sometimes it goes out and this re-lights and other times it changes color

cp2004, 2020-11-20:
Do you have a common ground between the strip, power supply and the raspberry pi? It sounds like this connection might be a bit loose, since that causes it to change colour and flicker often.

bouba830, 2020-11-20:
sorry, i don't speak english and the translation is poor lol, can you tell me that again with another word please, sorry again

cp2004, 2020-11-20:
Sorry, I will try and be clearer:

You need a ground connection (negative) between the power supply and the Raspberry Pi.

If this connection is bad, then you get flickering and bad colours.

Look at the pictures in the instructions, check you have the wiring right, link is in homepage.

bouba830, 2020-11-20:
it is true that for the tests I have a makeshift assembly, but it seems to me that the assembly is correct.
I will check it again, thank you very much

cp2004, 2021-01-26:
@bouba830 did you make any progress here?

cp2004, 2021-03-13:
If you are still having issues, please get back in touch. Since this issue, there has been a number of guides published on the documentation websites for solving issues, which may be of use for you.

closed 2021-03-13 by cp2004