cp2004/OctoPrint-WS281x_LED_Status #57: [Feature Request] Enhancement to progress effect for printers with LED strip on Z (sides) and X (top) of printer

james-fry, 2020-11-19:
I have an i3 style printer with a frame surrounding the print bed.
I have installed a strip of ws2812 leds around three sides of that frame - up one side of the frame, across the top of the frame, and back down the other side.
This means the progress effect tracks up then across then back down around the frame - which limits the ability to check the approx progress at a glance.

It would be great if its possible for the print progress effect to be enhanced toincrement from both ends of the strip so that the progress "bar chart" is on both sides of my printer with 0% at the bottom and 100% at the top.

Even better would be to be able to define segments for sides and top of printer and use these in different ways.
e.g. solid white light from top LEDs to illuminate the print bed, and progress tracked up both sides.

james-fry, 2020-11-19:
Current behaviour:
[image]

james-fry, 2020-11-19:
Proposed enhancement:
[image]

cp2004, 2020-11-19:
Its on my list to create different progress effects that may look better on your machine (for example, build from both ends).

I don't currently have any plans to implement customisable segments whilst I am aware that people would *like* that, it is quite a big project to implement. So I can say yes to some improvements, but not quite different zones *just yet*. Maybe in the further future as this project develops.

cp2004, 2021-03-01:
I have implemented a progress effect that runs 'from both ends', ready for 0.7.0 - expect a release candidate some point in the next week.

commit bd0fc01c (2021-03-01): :sparkles: New effect - print progress from both ends

Closes #57
closed 2021-03-11 by cp2004