cp2004/OctoPrint-WS281x_LED_Status #58: cooling progress effect

Bimb0, 2020-11-20:
The cooling progress effect only monitors the instrument temperature, even if I have it set to monitor the bed. Monitoring ends when the tool temperature is reached.

OctoPrint 1.4.2
OctoPI 0.18
Python 3.7.3

cp2004, 2020-11-20:
Can confirm, this does not appear to work.

cp2004, 2021-01-15:
Fixed for 0.7.0, it was simply a UI error where there was a typo in the setting.

cp2004, 2021-03-05:
Hey @Bimb0, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

cp2004, 2021-03-11:
0.7.0 has been released.

closed 2021-03-11 by cp2004