cp2004/OctoPrint-WS281x_LED_Status #59: [Feature Request] Linear-gradient effect

ShagoY, 2020-11-22:
Hi,

I use an [Adafruit NeoPixel Ring](https://www.adafruit.com/product/1463) (WS2812) that works perfectly with your plugin, my concern is that most of the effects are not useful because they are not visible enough.

I also use the progression effects and unfortunately the only one proposed (by the way, do you plan to add more?) is not enough.
Changing only one led among the 16 available during the printing/heating/cooling progression is clearly not usable with a ring.

So I was wondering if you could add the linear-gradient effect ?
This would allow to change all the leds from one color to another and make the effect much more readable.

Thanks :)

cp2004, 2020-11-22:
> (by the way, do you plan to add more?)

I was waiting for this request.... It's already on my list to add some different progress effects, for the next one. At least I have an issue to refer to now!

Currently on the list are:
* Fade - like you are suggesting
* One where progress bar moves from both ends to middle
* Opposite to the above, middle to ends

Maybe I will come up with more, but that's all for now. Planned for next release.

ShagoY, 2020-11-22:
Good to hear it! Thank you :)

And sorry I don't have any other idea of effects as I only use a ring led. So as long as the effect doesn't include all the leds, I don't really have any use for it :)

Except maybe a gradient on several colors (more than 2 so).
Or pulse/bounce/blink effects on two colors (instead of clear).

cp2004, 2021-01-19:
Implemented via the above commits, ready for 0.7.0.

ShagoY, 2021-01-28:
Thank you ! I will test as soon 0.7.0 is released!

cp2004, 2021-01-28:
I'll be publishing a release candidate before then, will ping you when it's ready for testing.

cp2004, 2021-03-05:
Hey @ShagoY, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

commit eb9390b6 (2021-01-19): :sparkles: Add linear gradient effect

Fades from one colour (base) to next colour (colour).

Closes #59
closed 2021-03-11 by cp2004