cp2004/OctoPrint-WS281x_LED_Status #60: [Feature Request] Multiple Strip type

ShagoY, 2020-11-24:
Hi :)

After seeing pictures of your plugin with strip leds (I have a neopixel Ring), I feel like adding a strip in my configuration just for the effects :-D

Currently my neopixel is around my camera and it allows me to light up my printing during the night (And during my snapshoots).
So I'd like to continue using it with your plugin but I'd like to add another strip for the progression/heating/cooling/etc. effects...

So I miss the possibility to add several led configuration.

I guess it must be a huge development work but I was wondering if it was in your "todo" ?

Thanks again !

cp2004, 2020-11-24:
Due to a hardware limitation of the Raspberry Pi, you can't connect more than one LED Strip in a way that works with OctoPrint.

You can chain WS281x LEDs together, so you connect them in a row. Using some clever cabling you might be able to connect the two together. However the plugin will just recognise this as a single strip, so that's not a *great* solution. This part would be the massive development work, to allow users to define different segments to do different things. 

Maybe I will consider it at some point, but given I already have quite a lengthy TODO list (this is not my only plugin, and real life as well) its not on there for the moment. 

Before then however, I am in progress working on a plugin to connect to [WLED](https://github.com/aircoookie/wled) that could support multiple strips, so this may become the preference for this feature and I won't add it to this plugin.

No for now, but note that this decision may be revisited in the future.

ShagoY, 2020-11-24:
I don't know how Octoprint works, I just assumed it was only related to pin GPIOs.

So if I understand even if I integrate another plugin (like Enclosure) to manage another strip on another pin GPIO, it wouldn't work?

cp2004, 2020-11-24:
Enclosure is a special case, since it gets the user to do some additional install steps and is very limited in it's control functionality.

It uses `sudo` to run external scripts that run the LEDs for it. Whilst this works, it is very limited in functionality as it is effectively 'disconnected' from the enclosure plugin once it is run. iirc this method only supports two different pins, unsure if they can both be used at the same time.

This plugin uses the SPI interface of the Raspberry Pi, which doesn't require root to run so it can be used directly inside OctoPrint no problems. Issue with this, is the Pi only has one SPI pin - so you can only have one strip.

PlzDontKillDave, 2021-02-08:
One thing i can think of... would kinda be a chore programming... Is to set up a way where the first 12 pixels is one configuration (lights around the camera) and then the next 8 pixels would be the status pixels

cp2004, 2021-02-08:
@PlzDontKillDave - you're right it would be a lot of work. I do have ideas, to setup segment control like [WLED](https://github.com/Aircoookie/WLED) has but there's a massive amount of work stacked up in the pile before anything like this could come about.