cp2004/OctoPrint-WS281x_LED_Status #61: Plugin does not work anymore after Raspian update and upgrade

wkoetz, 2020-11-26:
I run the following commands on the SSH shell for upgrading everything to the newest version (26.11.2020):
sudo apt-get update
sudo apt-get upgrade
sudo apt-get autoremove
sudo reboot
After that, the plugin does not work anymore. I also uninstalled it in Octoprint and installed it new. Does not work. The OS Config Test shows all green. It seems to be a change in the Raspian operating system. Could you please check, what is going wrong?

cp2004, 2020-11-26:
> Could you please check, what is going wrong?

Not easily...

That is probably out of my control, unless there is additional information in the logs. Please enable debug logging (Plugin settings > Features > Debug Logging), upload the `plugin_ws281x_led_status_debug.log` and `octoprint.log` so I can review.

Out of my control, since the LEDs are driven by a separate library. I can see if I can reproduce at some point, but not something I can fix unfortunately. If I can reproduce I will update here, and raise the ticket upstream.

cp2004, 2020-11-29:
Unable to reproduce. Updated my image here and it works just fine. 
@wkoetz I **need** more information about this report, or you will be forever stuck. If I don't get a response in a week I'll have to close this as unreproduced.

wkoetz, 2020-12-01:
When the printer powers up and octoprint starts, the LEDs turn to white and nothing happens with them anymore. See also the attached logs of a atarting print.
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5622982/plugin_ws281x_led_status_debug.log)
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5622983/octoprint.log)

cp2004, 2020-12-01:
I can't rule out that undervoltage is not the cause of this, since it works just fine for me on the same hardware & software versions.
>```

cp2004, 2020-12-01:
I cannot rule out that undervoltage is not the issue here, from your logs:
> ```
>2020-12-01 13:41:40,485 - octoprint.plugins.pi_support - WARNING - This Raspberry Pi is reporting problems that might lead ?to bad performance or errors caused by overheating or insufficient power.
>!!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.
>```

It works just fine for me on a fully updated instance too, and the exact same hardware model (Raspberry Pi 4B, 2GB/4GB, Rev 1.2). If you are still having issues once you have fixed the undervoltage issues, please post the full `/boot/config.txt` file here, for me to review.

wkoetz, 2020-12-01:
Unbelievable, with the original power supply it works again. Since more than a year I´ve powered my Raspi from the 5V from the printers mainboard, regardless of the undervoltage problem. Since a couple of month from the DCDC Power Module from the BTT mini E3 Board without having this problem. OK, I have to go now deeper into this power problem. Thanks a lot for your help!

closed 2020-12-01 by wkoetz

cp2004, 2020-12-01:
Good news! You had me panicking I was going to have to deep dive into the Raspberry Pi kernel, I was dreading having to deal with this issue.

wkoetz, 2020-12-01:
This behavior is just very strange, as it worked fine in the past. Something this last update caused the GPIO pin to stop working on undervoltage, which it didn't before. If I can't find a solution, I'll probably buy a buck converter for the Raspberry.