cp2004/OctoPrint-WS281x_LED_Status #62: Question: GPIO10 status on a RPI4 (multiple issues with RPI4 and the GPIO output)

MaikelKelder, 2020-12-01:
I ran successful this plugin on a RPI3B and still works.
Switching to a RPI4 with minimal difference, LED dont turn on.
My question is, in what status should GPIO10 be in?

RPI3B - GPIO10 - 0V (seems to be just signal)
RPI4 - GPIO10 - 3.3V (status MOSI UP)

I cant seem to get the SW2811 strip to work on a RPI4, all points to not getting any signal out of GPIO10.
I tried 2 RPI4's both the same.

Help is appreciated if there are any ideas.

cp2004, 2020-12-02:
Please post the log, and what does the 'OS Config text' report please.

cp2004, 2020-12-18:
@MaikelKelder any update on this issue?

The logs will help me determine if the issue is with the plugin, or potentially a hardware fault.

That's the `octoprint.log` and the `plugin_ws281x_led_status_debug.log`, they would both be very useful.

MaikelKelder, 2020-12-18:
[octoprint (2).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5717901/octoprint.2.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5717902/plugin_ws281x_led_status_debug.log)

?Sorry, took off my whole config, but build it up again after your message. Clean install. Uploaded the logs.

MaikelKelder, 2020-12-18:
Finding more people with the same error as in the logs:
https://github.com/jgarff/rpi_ws281x/issues/384

ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)

cp2004, 2020-12-18:
Try SSH to your Pi, run `~/oprint/bin/pip install rpi_ws281x==4.2.5`, then restart OctoPrint and see if it works now? They added support for this board rev. but it was after the latest update of my plugin, so hasn't yet been updated.

MaikelKelder, 2020-12-18:
YES! working!! Thanks for the support, quick fix!

closed 2020-12-18 by MaikelKelder

cp2004, 2020-12-18:
Excellent, I might see if I can push out an update soon, there's a couple of bugs that need fixing & that dependency will need to be updated.