cp2004/OctoPrint-WS281x_LED_Status #63: LED Strip on other GPIO port

Cl3tus, 2020-12-02:
Hello!

New to 3D printing, i stumbled upon your great plugin and it worked great till i got my PiTFT+ hooked up to it.

I have to choose a different GPIO pin now, i made myself root and added the --iknowwhatimdoing switch to the daemon args.

I tried GPIO22 (Pin 15) but no luck. Do i need another special GPIO or because of the PiTFT's nature of connecting through SPI is blocking this? If so, is there another way to solve it?

Thanks!

Regards, Paul

cp2004, 2020-12-02:
According to the library's docs, you can use:
> PWM0, which can be set to use GPIOs 12, 18, 40, and 52.
    Only 12 (pin 32) and 18 (pin 12) are available on the B+/2B/3B

> PWM1 which can be set to use GPIOs 13, 19, 41, 45 and 53.
    Only 13 is available on the B+/2B/PiZero/3B, on pin 33

So try using one of them.

 I should probably link it in the docs, under the [LED config stuff](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/LED-Strip-configuration#gpio-pin) but since I am not supporting it I didn't.

Cl3tus, 2020-12-03:
Thanks for your quick answer!

I've changed the pin to pin 33 (GPIO 13) and changed the settings accordingly.

[image]

I got a white light, and now its doing a green running bar so to say. Probably the green operational effect or something.

But when i try to test the colours in the menu it does not work. Is there anything else that needs to be done?

Thanks again!

cp2004, 2020-12-03:
The test requires the printer to be connected (yes that's not obvious) because I took a shortcut when making it. Will be improved for the next release, there's an issue about it somewhere.

Have fun with your LEDs!

closed 2020-12-03 by cp2004

Cl3tus, 2020-12-03:
Confirmed working! Thanks! Maybe for the picture if other people are trying the same.

Hardware:
Raspberry Pi 3 B+
Adafruit PiTFT+ 3.5"
SK6812 Strip

Wiring:
5v LED strip on Pin 2
Gnd LED strip on Pin 6
Data LED strip on Pin 33 (GPIO 13 - PWM1)

Set settings like above in the screenshot and good to go!

Indeed the issue can be closed! Thanks for your very quick feedback!