cp2004/OctoPrint-WS281x_LED_Status #64: Cant seem to get the plugin working i have looked through my log this is all i see...

iKonTechDev, 2020-12-06:
Not sure whats going on here but i cant seem to get this plugin to work on this PI.... Sharing GPIO with Touchscreen do i need to do anything special? 

2020-12-06 07:03:43,951 - octoprint.plugin - ERROR - Error while calling plugin ws281x_led_status
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_ws281x_led_status/__init__.py", line 605, in on_event
    self.update_effect(self.supported_events[event])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_ws281x_led_status/__init__.py", line 560, in update_effect
    if not self.SETTINGS[mode_name][
KeyError: u'idle'

cp2004, 2020-12-06:
> Sharing GPIO with Touchscreen

This may be the issue. Do you have a link to the touchscreen model? The plugin requires exclusive access to the SPI interface, and I am aware of touchscreens breaking things, see [this section of the troubleshooting guide.](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Troubleshooting-Guide#elecrow-5-inch-hdmi-touchscreen)

And could you look for the `plugin_ws281x_led_status_debug.log` & upload it, it will tell you if the LEDs *should*  be working or if it is having trouble getting set up.

iKonTechDev, 2020-12-06:
here is link to screen: https://www.adafruit.com/product/2441

Ya i guess your right how can we use a different GPIO i dont care about root but i was having an issue running as root as well

[2020-12-06 02:06:03,555] INFO: Initialising LED strip
[2020-12-06 02:06:03,559] ERROR: Strip failed to initialize, no effects will be run.
[2020-12-06 02:06:03,560] ERROR: Please check your settings.
[2020-12-06 02:06:03,560] ERROR: Here's the exception: ws2811_init failed with code -13 (Unable to initialize SPI)

Any ideas thanks that kind of sucks :(

iKonTechDev, 2020-12-06:
I thought that may be the problem but i disabled the plugin and no go the funnie thing before i ripped the printer apart to do SKR Mini update a bigtechtree TFT display and the BLTouch i used another raspberry and just installed everything and the leds would work so i know my wiring is right i used buck converter from main supply to level shifter to pi and leds strip like so

So with disabling plugin should of it worked??? because it did not... Also whats that deal with GPIO Master and Slzve hand off i thought there was a way to share a GPIO but i do notice the LEDS do some diffrent things when i touch the screen LOL

[image]
[image]

cp2004, 2020-12-06:
That's probably not the plugin, but rather the signals to the screen being picked up by the LEDs & interpreted in their own way.

iKonTechDev, 2020-12-06:
Ya hehe

iKonTechDev, 2020-12-06:
And i meant i disabled the TouchUI plugin sorry

cp2004, 2020-12-06:
There's not a lot else, other than use one of the PWM pins (see #63 for details) - and note this does mean you need to run OctoPrint as root, which shouldn't be done.

In the first part of 2021 I am hoping to release a plugin that connects [WLED](https://github.com/Aircoookie/WLED) devices over wifi, which would work for you in this case, but obviously that takes a fair bit of time & resources to do.

iKonTechDev, 2020-12-06:
Yas i hear you whats the easiest way to run Octo as Root i been looking not finding much about it

Thanks

iKonTechDev, 2020-12-06:
Think this would work: for any one else

I did get octoprint running as root though:

mounted the octopi sd card (on another pi or any linux machine)
edited the /etc/sudoers file (passwordless sudo)
pi ALL=(ALL:ALL) NOPASSWD:ALL
edited the /etc/default/octoprint file:
OCTOPRINT_USER=root
DAEMON_ARGS=--iknowwhatimdoing

cp2004, 2020-12-06:
Cool, if you have any more problems let me know.

closed 2020-12-06 by cp2004

iKonTechDev, 2020-12-06:
Ya word up all working!!! Thanks!