cp2004/OctoPrint-WS281x_LED_Status #65: [Feature Request] Retain 'power state' over reboot

Gordontgopher, 2020-12-07:
**Is your feature request related to a problem? Please describe.**

When I'm not printing I turn the plugin off. If I have to restart OctoPrint (plugin update or similar) then the LEDs are turned back on. It would be useful of the on/off state persisted over an OctoPrint (or Pi) restart to keep the LEDs in the preferred state.

**Describe the solution you'd like**

See above :)

**Describe alternatives you've considered**

Turning off some of the effects.

**Additional context**

cp2004, 2020-12-07:
Haha, this is something I noticed myself and intended to implement. Will do it for the next release, but hopefully not in a confusing way for people who don't want this. No ETA, since there are a lot of other projects in the backlog.

cp2004, 2021-03-01:
This is implemented by the above commit, ready for 0.7.0

cp2004, 2021-03-05:
Hey @Gordontgopher, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

Gordontgopher, 2021-03-06:
Thanks! Can confirm this feature works :)

commit 3b92ea7a (2021-03-01): :sparkles: Store lights on/off state across restarts

QoL at it's finest

Closes #65
closed 2021-03-06 by Gordontgopher