cp2004/OctoPrint-WS281x_LED_Status #66: LEDs always white?

jimpearceuk, 2020-12-13:
Hi guys, thanks for this plugin!

I have setup the strip and trying all the different types I always end up with white leds on. One setting was selected is good until I turn them on with the torch icon, then I cannot turn them off. 

I have managed to find one WS2811_STRIP_GRB which the colours are correct, if I could control the white I'd be there.

Appreciate any pointers please?

Thanks Jim

cp2004, 2020-12-16:
Hey @jimpearceuk - apologies for the late reply, I have been busy with other projects recently.

Do you have 'toggle' or 'timed' mode for the torch set in the settings? The torch effect takes maximum priority if I remember correctly, so until that effect ends then it will not change.

Can you enable 'debug logging' (under the features tab), then upload both the `plugin_ws281x_led_status_debug.log` file & the `octoprint.log` file from when you can reproduce this issue? I am wondering if there may be a bug that needs a fix hidden here.

Jester4977, 2020-12-18:
I'm having the same issue.  When I go into the plugin settings the lights flicker, then when I save and turn on the torch for example the lights work for about 1 second and then they stay solid white.  They wont change at all, or even turn off, until I change something in the settings.   Even trying to use the Test Red/Green/Blue buttons nothing happens, all the LED's stay white. 

I've run the OS Config test and everything is good there.  Double checked my common ground and overall wiring is correctly connected.  I tried reducing the voltage from 5 to 3.5 and nothing changed.

I even tried uninstalling and reinstalling the plugin and went through all the testing again, no dice.  I've tried every strip type, restarting Octoprint after saving, and that didn't work either.

**Octoprint.log**
2020-12-17 21:24:51,890 - octoprint.plugins.ws281x_led_status - INFO - Running OS config test (UI Mode)
2020-12-17 21:24:54,496 - octoprint.plugins.ws281x_led_status - INFO - OS config test complete. Results:
| - set_core_freq: passed
| - spi_enabled: passed
| - adduser: passed
| - spi_buffer_increase: passed
| - set_core_freq_min: passed
2020-12-17 21:25:17,050 - octoprint.plugins.ws281x_led_status - INFO - Settings refreshed
2020-12-17 21:25:17,186 - octoprint.plugins.ws281x_led_status - INFO - WS281x LED Status runner stopped
2020-12-17 21:25:17,187 - octoprint.plugins.ws281x_led_status - INFO - WS281x LED Status runner stopped
2020-12-17 21:25:17,201 - octoprint.plugins.ws281x_led_status - INFO - Ws281x LED Status runner started
2020-12-17 21:25:23,654 - octoprint.plugins.ws281x_led_status - INFO - Turning light off
2020-12-17 21:26:24,482 - octoprint.plugins.ws281x_led_status - INFO - Turning lights on

**plugin_ws281x_led_status_debug.log**
[2020-12-17 21:25:17,170] INFO: Kill message recieved, Bye!
[2020-12-17 21:25:17,232] INFO: Initialising LED strip
[2020-12-17 21:25:17,238] INFO: Strip successfully initialised
[2020-12-17 21:25:17,242] DEBUG: Current settings:
 | * STRIP SETTINGS *
 | - led_pin: 10
 | - reverse: False
 | - led_invert: False
 | - strip_type: WS2811_STRIP_GRB
 | - led_dma: 10
 | - led_channel: 0
 | - led_count: 22
 | - led_brightness: 115
 | - led_freq_hz: 800000
 | * EFFECT SETTINGS *
 | torch
 | - color: #ffffff
 | - delay: 1.0
 | - enabled: True
 | - effect: cycle
 | disconnected
 | - color: #000000
 | - delay: 25.0
 | - enabled: True
 | - effect: cycle
 | success
 | - color: #000000
 | - delay: 25.0
 | - enabled: True
 | - effect: rainbow
 | failed
 | - color: #ff0000
 | - delay: 10.0
 | - enabled: True
 | - effect: pulse
 | startup
 | - color: #00ff00
 | - delay: 75.0
 | - enabled: True
 | - effect: wipe
 | progress_print
 | - color: #00ff00
 | - base: #000000
 | - enabled: True
 | paused
 | - color: #0000ff
 | - delay: 40.0
 | - enabled: True
 | - effect: bounce
 | progress_cooling
 | - color: #ff0000
 | - base: #0000ff
 | - enabled: True
 | idle
 | - color: #00ccf0
 | - delay: 75.0
 | - enabled: True
 | - effect: wipe2
 | printing
 | - color: #ffffff
 | - delay: 1.0
 | - enabled: False
 | - effect: solid
 | progress_heatup
 | - color: #ff0000
 | - base: #0000ff
 | - enabled: True
 | * ACTIVE TIMES *
 | - start: None
 | - end: None
[2020-12-17 21:25:17,243] INFO: On message recieved, turning on LEDs
[2020-12-17 21:25:23,666] INFO: Off message recieved, turning off LEDs
[2020-12-17 21:26:22,280] INFO: Off message recieved, turning off LEDs
[2020-12-17 21:26:24,526] INFO: On message recieved, turning on LEDs
[2020-12-17 21:26:29,105] DEBUG: Recieved message to change effect: torch

Jester4977, 2020-12-18:
I may have found a solution.  Something bingston said in his last comment in help #67 was that he tried setting the Hz to 400000 and the lights went white.  I tried this and mine went white as well and they no longer flickered the correct color for 1 second when entering the settings menu.  So I changed the Hz to 1200000 and now it seems to be working.  I can turn the lights on/off with the toggle, and so far the other settings are working as well, including the test Red/Green/Blue buttons. 

Will post back with updates should anything change.
Thanks-
John

Jester4977, 2020-12-18:
After further testing I have successfully changed the frequency from 1200000 to 1400000 and the lights still worked.  I pushed it even farther up to 1800000 but that started causing some of the lights to go white again, so I backed it down to 1250000.  All features seem to be working great at this frequency.

jimpearceuk, 2020-12-18:
Thanks a million guys I will test tomorrow and advise, cheers for all your help!

cp2004, 2020-12-18:
@Jester4977 That's interesting - I wasn't aware that there were strip models out there that required different frequencies. I guess that's why the option was there 🤷 

I know there were some older strips that used 400K, instead of 800k but didn't know they went higher. If I can work out a way of phrasing it, I'll add it to the documentation and/or troubleshooting guide.

jimpearceuk, 2020-12-18:
Hey guys, managed to find time to test it, works a charm thanks!!

cp2004, 2020-12-18:
Cool, thanks for letting me know the solution! Will add it to the docs over the weekend probably. I find it a strange coincidence that no one has had this issue, then two come along at once - maybe there is a new strip variant being solved.

closed 2020-12-18 by cp2004