cp2004/OctoPrint-WS281x_LED_Status #67: Possible help

bingston, 2020-12-13:
I'm fairly new to printing but I decided to upgrade my Ender 3 Pro to an SKR mini E3 v2.  In doing so I noticed that there was a spot on the motherboard labeled Neopixels.  I was never able to get them working, got frustrated and while looking for a solution I stumbled across your plugin and thought it was a great idea.  I finally got around to trying to hook it up.  
I have a separate 5v source going to the lights and the lights share a ground with the pi.  The lights just flicker.  At first they stayed green and flickered, not they're more white but I can't them to stop flashing.  At one point it moved to more of a pulsing, but the flashing never stopped.  
[This](https://www.amazon.com/Adafruit-NeoPixel-Arduino-Integrated-Drivers/dp/B00IEDH26K/ref=sr_1_1?dchild=1&keywords=Adafruit+NeoPixel+Stick+for+Arduino-+8+x+WS2812+5050+RGB+LED+with+Integrated+Drivers+Brand%3A+Adafruit&qid=1607893331&s=industrial&sr=1-1) is the light strip that I got .  I don't know if it's different because it's an Adafruit, but I'd love to find a way to make this work.  Any chance this will work with your plugin? 
Thanks in advance!

cp2004, 2020-12-16:
Hey @bingston - sorry for the late reply, I have been busy with other projects over the last couple of days.

This kind of flickering is not quite what I have seen before, so I am not yet sure of the problem. It looks like the power supply may not be providing the correct power, since usually when it is the signal side (eg, no common grounds) they flash lots of random colours or just white.

What spec (do you have a link?) power supply are you using for these LEDs?

bingston, 2020-12-16:
Hi @cp2004!  Thanks for getting back to me at all.  I thought this was a great alternative to getting it to work with the SKR Mini E3 v2 (I've pretty much given up on that), so thanks for taking the time to make this.  I'm using an old wall wart I found laying around the house.  It's 5V 2500mA so I thought it would be enough to power these.  Not sure if it's needs some kind of special library because they're Adafruit - that was the issue trying to get it to work with the SKR Mini, if I understand correctly.  I can try another power source, I just thought that this one would have been enough.

cp2004, 2020-12-17:
The Adafruit LEDs are directly supported by this plugin, since they popularised the use of WS2811b, WS2812 and SK6812 LEDs under the name 'neopixel' - same LED, more expensive but probably higher quality service than direct out of China as I have got a lot of mine.

5V 2.5A *should* be enough, but not all power supplies are created equally and some can only sustain that kind of current for short periods of time, which is what I can see from your video. Obviously without access to your setup I can't be sure, but if you have another PSU then it might be worth trying it.

bingston, 2020-12-17:
Thanks for getting back to me again.  Happy to hear that neopixels is supported.  Like I said, I initially bought this small strip to plug directly into the SKR Mini, but since I can't get that to work because of library issues, I'd like to use this strip since I bought it.  I'll try changing the power supply to see if that helps.  I think I have some other lights from previous projects laying around too that I can try - I would just need a strip that has power, ground, and data right?  
Thanks again for the tips!  I'll report back if I make any progress in case it can help anyone else.

cp2004, 2020-12-17:
Yes any of these strip types are supported:
> WS2812
WS2811 GRB
WS2811 RGB
WS2811 RBG
WS2811 GBR
WS2811 BGR
WS2811 BRG
SK6812
SK6812 W
SK6812 RGBW
SK6812 RBGW
SK6812 GRBW
SK6812 GBRW
SK6812 BRGW
SK6812 BGRW

https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/LED-Strip-configuration#strip-type

The wiring diagrams you have probably already seen, they show the pins needed:

https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Wiring-your-LEDs

bingston, 2020-12-17:
Well, I have tried two different power supplies, and two different kinds of lights and can't stop it from blinking.  I don't know if this is a factor, but I have one raspberry pi that controls two different printers (have to use a different port number to use the other instance of octoprint).  Not sure if that could be messing anything up but wanted to mention that.
I've uninstalled the plugin and tried reinstalling it with the same results.  I've also checked the voltage at the end of the strip and it's still at 5v, so I think the power supply is ok.
If you have any other suggestions, I'm happy to try them.  It **must** be something **I'm** doing wrong because others have gotten it to work, but I can't seem to figure it out

cp2004, 2020-12-18:
I don't really have much more of an idea, you could try and take the LED strip out of the plugin/Pi and test it with a different program, see if there are any different settings.

In #66 they mentioned increasing the clock, but then that was because the LEDs were white instead of flickering as they are for you.

Do you have anything else installed on the Pi? I am at a loss as to what could be making them flicker, this I haven't seen before. And I've seen quite a few issues around here. 

I'll continue to think, and if I have any other ideas I'll get back to you.

bingston, 2020-12-18:
Thanks for getting back to me.  I only have octoprint installed on that pi, but as I mentioned I followed a tutorial to get it to control two printers.  That's the only weird thing I can think of.  I disabled all of my plugins in case there was a conflict there too, but didn't have any luck.  I'd love to get it working, but I'm at a loss.  Thanks for trying to help me out though, I appreciate it!

cp2004, 2021-05-08:
@bingston - it's been a while since this issue had any updates. Is this still an issue? Did you get it resolved?

Since this was posted, I've been keeping a list of issues in the troubleshooting guide and re-wrote all of the documentation to provide more help. It's reduced a number of the support questions I got here, so I was wondering if there was anything there that may help you?

https://cp2004.gitbook.io/ws281x-led-status/troubleshooting-guide

bingston, 2021-05-08:
@cp2004 - I never got it working.  I didn't want to keep bugging you and I wasn't able to get it figured out.  I've been trying to get the Neopixel port to work with the new Marlin 2.0.8, but I've been struggling with that too.  Now that there's an update for this, I'll try to revisit the installation and see if I can get it working.  I'll report back to let you know how it goes.  Thanks for the update!

cp2004, 2021-05-08:
If you feel like trying again, then be sure to ask questions! Other people occasionally have better ideas on the OctoPrint discord server too, I'm often around there. I still occasionally learn new things helping people out, so here's a list of some new ideas:

* Make sure you keep the wires to the first LEDs as short as possible, to reduce voltage drop. You can also use a single sacrificial pixel (guide linked in troubleshooting guide)
* If you can, try and adjust the LED strip voltage up or down. Some LEDs are not as tolerant as others.

closed 2021-08-25 by github-actions[bot]