cp2004/OctoPrint-WS281x_LED_Status #68: GPIO With 3.5" Touchscreen

csteamengine, 2020-12-17:
Hello, I recently setup an enclosure with LEDs using the Enclosure Plugin. I decided to add an OctoScreen interface to my enclosure so I didn't have to go to a computer to start/stop prints anymore. Anyways, with the addition of the touchscreen which doesn't use ALL of the GPIO pins, I had to switch the LEDs to pin 21, but in the end they are working. 

I then came across this plugin and it seems much more robust, active, and feature-filled, but I can't get it to work with my current setup. I can confirm that the settings for strip type, pin, etc... are all correct and I have run the setup wizard completely. However, when I use the utilities to test the LED colors, nothing happens. 

I know in the docs you mention that only pin 10 can be used, but as I have already proven it is possible to use pin 21 using a different plugin, is there any chance you could help me get your plugin working? 

Any help you could give would be greatly appreciated!

[Enclosure Plugin I'm Currently Using](https://plugins.octoprint.org/plugins/enclosure/)
[LEDs I'm Using](https://www.amazon.com/BTF-LIGHTING-Flexible-Individually-Addressable-Non-waterproof/dp/B01CDTEJBG/ref=sr_1_1_sspa?dchild=1&keywords=ws2812b&qid=1608209133&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFRTFlFQTZYRzZCUDImZW5jcnlwdGVkSWQ9QTA4MzM3MDdWQTVWRU5NUFdHWDEmZW5jcnlwdGVkQWRJZD1BMTAwMzgzNDE0Uzc0STZMSko4MFYmd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl)

cp2004, 2020-12-17:
The way the enclosure plugin works is by using `sudo` to run an external script. With the way this plugin works to enable far better features is by controlling the LEDs directly from within OctoPrint, however this has the limitation of pin 10 since it is the only one that works without root access.

You can use the other pins, but you will have to run OctoPrint as root which is really not recommended (running a webserver that has root access is unsafe, make sure you don't get hacked 😉). It is technically supported in the plugin, see #63 and #64 for similar cases. I should probably update the docs to include this, since people keep asking...

If you open the `plugin_ws281x_led_status.log` you will probably find some problems reported about permission denied or something.

csteamengine, 2020-12-17:
Thanks so much for the quick feedback! It is unfortunate that pin 10 is the only one that can be used, since I'd like to have your plugin AND a touchscreen, but I guess I'm getting greedy at this point 😆. 

I seriously want to say good job on this plugin it is beautifully done, makes me miss my programming days! I run a YouTube channel called Modern Hobbyist and I'll definitely be making a video about this plugin!

Cheers!

cp2004, 2020-12-17:
Thanks for the feedback 🙂 

Yes it is a trade off, LEDs vs. touchscreen. I have a plugin for WLED in the pipeline, that removes the GPIO need entirely and will connect to a [WLED](https://github.com/Aircoookie/wled) instance for wireless LED control. That will take a while though, as this one has done over the last few months. I look forward to watching the video 🙂

closed 2020-12-17 by csteamengine