cp2004/OctoPrint-WS281x_LED_Status #70: [Feature Request] Single LED for printing progress

Jester4977, 2020-12-18:
It would be great if you could set a single LED to be lit with the selected color instead of the entire left side of the bar.  For example, if your print is halfway completed, just light up the LED in the center of the bar green, and the rest of the LEDs can remain white.  Using the majority of the LEDs white helps with timelapse videos, and having a single LED moving from left to right can provide the progress without disrupting the color for the pictures/video.

cp2004, 2020-12-18:
Similarly to #57 and #59, I'll be adding a wide range of progress style effects in 0.7.0. When this will be, who knows, but at some point it will arrive.

cp2004, 2021-01-19:
Implemented via the above commits, ready for 0.7.0.

cp2004, 2021-03-05:
Hey @Jester4977, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

commit f4515ac2 (2021-01-19): :sparkles: Add single pixel progress effect

Closes #70
closed 2021-03-11 by cp2004