cp2004/OctoPrint-WS281x_LED_Status #71: [Bug] Dont Work with Docker

FDGod07, 2020-12-22:
**Description of the bug**
Cant install it on Octoprint

**Steps to reproduce**
Install Ubuntu 20.04 on Pi4
Install Docker
Install Octoprint via Docker
Try to Install the Plugin

**What did you expect to happen**
Shows up after Installation

**Logs and screenshots**
[octoprint (1).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5732345/octoprint.1.log)
Raspberry Pi 4 Model B Rev 1.2

**Version of plugin and OctoPrint**
OctoPrint: 1.5.2
WS281x LED Status  0.6.0

cp2004, 2020-12-22:
From discussion on discord, it seems that from within docker the path to `/proc/device-tree/cpuinfo` does not work.

Needs a way of forcing through the check & a different way of determining the Pi model.

cp2004, 2020-12-22:
Will try and implement it, but it's holiday season & I don't know when I'll get a chance to publish an update. Will keep you posted.

cp2004, 2021-02-05:
Hey @FDGod07!

I looked into this today, and managed to get it working. I've published a guide describing how it can be done at the link below:

https://cp2004.gitbook.io/ws281x-led-status/guides/setup-in-docker

Hopefully this works for you, and we can close this issue.

cp2004, 2021-02-13:
I'm going to close this issue now, feel free to give the above guide a go to get your LEDs working on docker. The documentation is in place and everything works just fine.

closed 2021-02-13 by cp2004