cp2004/OctoPrint-WS281x_LED_Status #72: Occasional flashing of one or two pixel elements during startup wipe sequence

gnattycole, 2020-12-29:
**What's your question?**Hi, just installed this on my Predator with a 40 element LED Neopixel ring, mounted in the roof of the build area. Looks really cool and initial startup seemed to go fine. I changed the colour of the startup wipe from the default green to a shade of blue and ever since, I get rando flashes from the LED's, often pixel element No:1, usually briefly flashing red, then occasionally I'll get the first quarter of the ring flicker around twice the brightness of the lit wiping LED's. The printer is idle so don't think it's interference from anywhere. I'm running the Pi from a 24 volt to USB buck converter and the Neopixel ring is powered from a dedicated 10 Amp 24 volt - 5volt buck converter. I'm not using any logic voltage converters although I've ordered a couple just in case it's this but when I had them working on the bench in the current config they seemed pretty solid. The ground of the LED and buck converter and pin 3 on the GPIO are linked. GPIO 10 is connected to the LED SIG-IN wire and the 5V-+ve is connected to the Buck converter positive. 
  I've tried using the solid colours in the utilities tabs which is successful, but still get the odd flash or flicker. Green seems to be the most stable with Blue the worst.
 Not sure if this is something stupid. Any info appreciated.

Regards, Simon.

gnattycole, 2020-12-30:
Update on this. After many hours of trying to diagnose this, I'm still having the same issues. Tried fitting a diode in the 5 volt +ve to the LED strip,, made it worse. I swapped out the 10 amp 5volt buck converter to an adjustable buck converter and played around with the output voltages between 3 volts and 5.2 volts. I didn't want to go any higher for fear of frying the led ring, - made it worse. I've now fitted a logic level shifter which has sort of stabilised it a bit but still getting random flashes of different colours from the ring and the first two LED's are now permanently white, 1 flashing at around 5Htz and the other on steady. The ring is performing otherwise as expected with regards to wipe, bounce etc.

gnattycole, 2020-12-30:
OK, think I might have got to the bottom of this now. After all the other trials, I decided to try an external mains phone charger to power the Raspberry Pi instead of the dedicated buck converter and it stopped all the flashing and misbehaving. So, I disconnected it again, took and old USB cable and cut the end off and connected the red and black wires to the 5 volts high current Buck converter powering the Neopixels. An it all works now. Well, all except the first LED in the chain which glows green when the rest of the LED are Blue. When in the "Wipe" sequence, it glows green for the first pass, then white for the 2nd pass. Otherwise it's working fine. Now I've just got to work out how to configure it to what I want.

cp2004, 2020-12-30:
Ah, I was just reading this and was preparing to reply, when I saw your second message! I was thinking it might have been some bad connection on the ground, either in the power supply or somewhere along the lines. These LEDs can be quite sensitive to anything that is not 'clean' power so to speak. I've made some dodgy connections before, where I could touch it, it would stop working.

Sounds like you got it all worked out. I've seen that 1st LED green issue before, let me find the ticket... #24 is the one. Said the logic converter was not playing nice. I've never seen that one *personally*, so can only relay what I've seen about.

Enjoy the plugin, you might be able to skip the first LED by changing the wiring (if you're feeling creative!) or cutting it out. No guarantees that it is the hardware itself, it is more likely the signal is a *little* off. Or there's a ghost about. You never quite know...

gnattycole, 2020-12-31:
Hi Charlie,  Thanks for the reply. Yeah, the green led was probably present on the bench but because the default startup setting was green wipe, I wouldn't have noticed.  I'll give you a shout if I find something but the ring itself I think is OK as it will change colour under certain conditions.Thanks for your work on this project. Great stuff. One suggestion if its possible would be a test button on each of the features, just to see what the effect will look like when you're changing settings.Best regards, Simon.Sent from my Galaxy

cp2004, 2021-02-08:
@gnattycole can I close this now? Did you get everything sorted?

WRT the suggestion in your final comment that I seem to have missed, it is in the back of my brain to look at - the best I can do at the moment is link the [effects documentation](https://cp2004.gitbook.io/ws281x-led-status/configuration/printing-effects) which will give you an idea of what each effect will do. Maybe it will turn up in a future release 🙂

gnattycole, 2021-02-08:
Hi, Yeah, no problem. I've managed to calm down the erratic behaviour by fitting a 470 ohm resistor in series with the signal pin.   Weird but worked.Best regards,Simon.Sent from my Galaxy

cp2004, 2021-02-08:
Thanks, I've added this to the Troubleshooting guide:

* https://cp2004.gitbook.io/ws281x-led-status/troubleshooting-guide#random-flickering-of-leds-unstable-signal

* Commit d0d0e68f7e92f38411c2d9a7a2f5f536f2ed08f9

closed 2021-02-08 by cp2004