cp2004/OctoPrint-WS281x_LED_Status #73: Update rpi_ws281x dependency needed

cp2004, 2021-01-01:
## The problem

Newer revisions of the Pi are not supported by 4.2.4, need 4.2.5 to work.

## The solution

* Update to 4.2.5 in this plugin
Also need to wait on *another* upstream pull request, https://github.com/jgarff/rpi_ws281x/pull/438 to add one more revision.

cp2004, 2021-01-01:
[rpi_ws281x-4.2.4.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5759213/rpi_ws281x-4.2.4.zip)

For anyone else who runs into issues with unsupported boards, download the above file and head to the plugin manager > get more > from uploaded archive. Then restart OctoPrint and you should be OK.

Once these changes are merged upstream, I'll be able to pull in the dependency properly. Or figure out a way of specifying the file in `setup.py`.

Tooldad, 2021-02-11:
Hi Charlie.  My LEDs are not lighting up at all.  I'm thinking I may be having the problem listed here since I am using a PI 4 B.  I tried to follow the directions listed (to upload new zip version), but I don't have the option to "upload from archive".  I only have the option to upload from URL or from a single file.  The single file option doesn't take a zip file.  Any suggestions?

cp2004, 2021-02-11:
You could try this URL in the relevant box instead:
```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5759213/rpi_ws281x-4.2.4.zip
```

Tooldad, 2021-02-11:
I should have added this from my log.....

(Hardware revision is not supported)
[2021-02-11 18:16:53,094] INFO: No strip initialised, exiting the effect process.
[2021-02-11 18:20:55,069] INFO: Initialising LED strip
[2021-02-11 18:20:55,071] ERROR: Strip failed to initialize, no effects will be run.
[2021-02-11 18:20:55,071] ERROR: Please check your settings.
[2021-02-11 18:20:55,072] ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)

Tooldad, 2021-02-11:
The URL worked.  Awesome.  Thank you sir!!

Josiahpox, 2021-03-03:
Hi, cp2004. I am still having some issues with my LEDs. I can get them to turn on, but there seems to be a problem when changing colours. It defaults automatically to white and I cannot get it to display any other colour unless I click the "save" button on the plugin, it then changes briefly to the specified colour, before changing back to white. 

Let me know if you need any more information.
Thanks in advance!

cp2004, 2021-03-11:
The version has been bumped to a minimum, still waiting on the rpi_ws281x update for the RPi 4B Rev 1.4 2GB (I think) and the compute module, but it will work for pretty much all of them now. And it will be automatically pulled in anyway now.

closed 2021-03-11 by cp2004

CommittotheIndian, 2021-04-02:
Based on what I've read, I think I'm running into this issue. I can not get past the SPI Buffer test. I am using 0.7.2 on a pi 4. 

I can not load the version via the URL mentioned higher in the threat. It claims to load and restart my octoprint but no plugin ever shows up in the plugin manager.

cp2004, 2021-04-02:
The SPI buffer test is unrelated to this issue - this would show up in the `plugin_ws281x_led_status_debug.log` file as Unsupported boards.

There should be no further issues with this and the link above is outdated. It's not a plugin either, just the dependency. If you have installed that it may well *now* be outdated and you will have to reinstall the plugin.

The SPI buffer test looks in the `/boot/cmdline.txt` file - if you are still struggling with this issue open an issue for support or get in contact on the [community forums](https://community.octoprint.org) or on the [discord server](https://discord.octoprint.org)

CommittotheIndian, 2021-04-02:
I'm sorry you responded just as I was about to delete. Apparently I can't count and I was off by a pin. Everything seems to be working fine. I'll delete if you like.

cp2004, 2021-04-02:
It's fine, enjoy the plugin 😄