cp2004/OctoPrint-WS281x_LED_Status #75: Can not control the LEDS

scr34m88, 2021-01-05:
Hey,

my problem is, that I cannot control the LEDs.
When Octoprint is booting up it's showing the Color Wipe effect. When Octoprint finished booting the LEDs are white.
If I go to the settings or reload the Webpage the LEDs will blank out and show up in white again.
If I try to use the test buttons in settings nothing happens...
I don't now whats wrong...

cp2004, 2021-01-05:
Please upload the log files, the `octoprint.log` and the `plugin_ws281x_led_status.log`, so I can see what's happening.

If you want, enable 'debug logging' in the features tab, and then test again and upload the files.

scr34m88, 2021-01-05:
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5772058/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5772060/plugin_ws281x_led_status_debug.log)

Here are the logfiles 👍 
Thank you for your really quick help!!

cp2004, 2021-01-05:
For the LED test to work, the printer (any will do, even virtual) must be connected. It's because I made them a lazy way, it will be fixed in the future.

The white LEDs are strange, since the setup passed all the tests, and there are no other effects played (in the debug log). Are you using an external power supply, and if yes (and you should) then the ground connections should be connected.

Hmm, it is hard to tell why they turn white on their own. The expected behaviour is that they briefly turn off when the settings are changed, so it can update them on the strip side of the plugin. The web interface loading shouldn't affect this, but please check about the torch mode - I'm not sure but want to rule it out, try changing the colour of it.

scr34m88, 2021-01-05:
Yes, I'm using an Powersupply for Raspberry and the LEDs the Ground wires are connected.
I'm using the WS2812b without an levelshifter.I think they arent RGB they are GRB is this a problem?  I tried a diode to get down the LED supply voltage, but the issues weren't solved.

I also changed the color of the torch.
Another issue is that during the bootup when the Color Wipe is showing, that sometimes all LEDs are flashing white for a short time and then the color wipe is going on.

Maybe this is an issue of not using an levelshifter?

cp2004, 2021-01-05:
Usually not using a level shifter results in no signal at all, or if it is on the edge, the effects may look as if they 'stutter'.

It sounds more like some loose connections, usually with flashing. I have a test setup here, and if I move the connectors around the LEDs can flash white similar to what you are seeing. Maybe that is the issue. Are they directly soldered or using a breadboard or something else?

scr34m88, 2021-01-05:
They are connected with wago clamps. I don't think that there are bad connections.
Maybe I got a wrong plugin setup?
Its also really strange that the light bulb button on the top has no function. I cannot turn the LEDs off...

cp2004, 2021-01-06:
I've googled the wago clamps, while they look quite solid I'm going to be honest I have no idea since I've never used one. This is what I am thinking so far, based on the info you've provided:
* The Raspberry Pi OS is setup correctly, confirmed by the logs (OS configuration test runs every boot of OctoPrint, in the logs)
* The power supply is ok, since if it was inadequate you might get the LEDs turning on/off flashing etc. 
* I am suspicious of the data connection. It sounds like there is either a lot of interference or resistance that means that not all the signals are getting through. What tends to be the case with WS281x is that when some data doesn't go through, you get sections with the wrong colour, or they turn white. This is from my experience working with these LEDs.
* You mentioned you think they are GRB not RGB leds. There are options for all sorts of combinations, you will probably say 'but only one for WS2812' but please try some other options. What they told you the LEDs were when you bought them is not necessarily the right option in the plugin.

That's my thoughts so far, I think to test this you could probably try different strip types (in the settings) or trying a different connection method - doesn't have to be permanent, just some experiments to see if you can change *anything* that might give us a clue as to what is affecting them.

scr34m88, 2021-01-06:
Okey thanks for your help.
Yesterday I tested my setup with nodemcu (esp8266) 3,3v logic. And there where no issues.
So I think there are no mechanical or electrical issues...

scr34m88, 2021-01-06:
I tried another led strip with only two leds and solderd every wire together. 
The wires are about 3-5cm long.

And there are the same problems. I think its a software issue....

scr34m88, 2021-01-06:
I meaured the data wire with an osciloscope. And I can see that the LEDs get an Signal for become white  . And during bootup the another signal for the color wipe effect. So the connection is good. The problem is that the Raspberry is telling shit to the leds

cp2004, 2021-01-06:
OK, that is interesting. I'm assuming you don't have anything else connected to the Pi that is interfering, so maybe there is something internally that is messing. I *do* know that when run over SPI we are at the mercy of the core clock frequency, which can vary. This is why there is a step of setting `core_freq_min=500` in config.txt - it is an SPI/WS281x compatible frequency. Not my suspicion of the data line is correct, but it is not the origin I thought... Something in the boot process is changing this frequency to one that is not supported by these LEDs.

We move on to that section of troubleshooting then I guess. (I apologise for the issues, it is difficult to debug without being there, seeing things).

Could you post the full contents of the `config.txt` file (located under `/boot/config.txt`, so you can get it when you take the SD card out if SSH is more difficult). I am looking to see if there are any conflicting settings that might be in there, the OS config test passed however it only picks up if the settings exist in the file, not that they are actually in operation. Other than that... We could try using PWM to control them, but this means you have to run OctoPrint as root which is not recommended or secure so that would be something you decide to try doing.

scr34m88, 2021-01-06:
[config.txt](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5776432/config.txt)
For sure here is the config file

The most people using ws2812b leds are use the gpio18 maybe we could try this...
But I don't know how to run octoprint as root maybe you can explain this to me.

cp2004, 2021-01-06:
GPIO 18 is the PWM option, but this requires root.

Try removing the line `core_freq=250` from config.txt

```diff
  dtparam=spi=on
- core_freq=250
  core_freq_min=500
```

Did you use a different model Pi (Raspberry Pi 3 or earlier) before you used the Pi 4? Since that `core_freq=250` line should only be there for a Pi 3. After all of this, maybe I need to add part of the OS configuration test to make sure that line is not there. As you might be able to tell, 500 being the minimum is not compatible with the base being 250....

scr34m88, 2021-01-06:
YES! Thats it!
It's working now! Thank you for your help and your time!

No I didn't used a Pi 3 before only my Pi 4...
So that was the Problem maybe you can add a check if this is Pi 4 that the line core_freq musn't be in the config. :)

cp2004, 2021-01-06:
Good news! Ok, leave this issue open and I will make sure it is in for the next release. There's a small stack of requests & a couple of bugs, for the 0.7.0 release.

Thanks for your patience, willingness to try things and clear descriptions of the problems! It made it a lot easier to understand.

kritch83, 2021-01-08:
FYI I also had this exact same issue today when using your plugin for the first time.  I'm using the 4b and same problem.  Neopixels show the correct color/pixel for a sec then go to all pixels, all white.  My config.txt had both core_freq & core_freq_min enabled.  I removed the one setting it to 250 and it solved my issue.  Thanks!

cp2004, 2021-01-08:
I hope there isn't a bug that causes that line to be put in there... will have to verify when I get back to working on this soon.

cp2004, 2021-02-13:
As of 0.7.0, the plugin will check that `core_freq=250` is not in `/boot/config.txt` and prompt the user to remove this manually in OS configuration test, which is also now integrated into the setup wizard.

mon5termatt, 2021-02-21:
So on my pi 3b Does it need the core_freq=250 because when its there it flickers. but when its not it blinks white

cp2004, 2021-02-21:
@mon5termatt a Pi 3, this line does have to be there, to get the clock to run at an SPI compatible speed. Leave the line in there, and try some of the other solutions to get the LEDs to stop flickering from the troubleshooting guide.

cp2004, 2021-03-05:
Hey @scr34m88, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

cp2004, 2021-03-11:
0.7.0 has been released.

closed 2021-03-11 by cp2004