cp2004/OctoPrint-WS281x_LED_Status #76: gpio pin 10

climatec, 2021-01-08:
Hi 
very nice add-on you have made.

but you say it not possible to have from gpio 10?  in the program its possible.

the problem is some of us is using a display that use first part of the gpio´s  pin 1 to 28.

any solution on that problem.

best regards

Henrik

cp2004, 2021-01-08:
See #63 for an explanation. GPIO pin 10 is the only pin that works without root, the rest you need to run OctoPrint as root to enable. This is not a limitation set by me, but by the hardware of the raspberry pi. Please note that choosing a pin other than GPIO 10 is not officially supported by me, due to the security issues of running a webserver as root. If you know what you are doing, it is possible.

climatec, 2021-01-08:
Hi thanks for replying so fast.

But as you can understand I can’t install the screen any different.  😢

So I have to abandon the idea. Unfortunately else it was a great idea.

Thanks.

Regards

Henrik

closed 2021-01-08 by cp2004